nordicsemi/Android-nRF-Mesh-Library #2: Provisioning fails for mynewt-based device

kabbi, 2018-06-09:
---

**Describe the bug**
Provisioning fails after `Sending public key` state.

**To Reproduce**
I'm using hardware mynewt-based mesh network node, on RedBear blenano2 (nrf52) device.
This same device is confirmed to work with SiliconLabs demo app.

1. Reset the device
2. Scan for devices in the app
3. Select the device and confirm provisioning
4. I see "Provisioning failed" message with the text of "App key was added successfully" (weird but true)

**Expected behavior**
Provisioning completes successfully.

**Platform details:**
 - Device: Nexus 5X
 - OS: Android 8.1
 - Library Version: [dev branch](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/commit/e556f9751420b878fd70921cacb293e62dde7b2c)

**Logs**
```
06-09 23:01:24.193 23694-23694/no.nordicsemi.android.nrfmeshprovisioner V/BleMeshManager: Sending data : 0x03000A
06-09 23:01:24.199 23694-23741/no.nordicsemi.android.nrfmeshprovisioner V/BleMeshManager: Data written: 0x03000A
06-09 23:01:24.199 23694-23741/no.nordicsemi.android.nrfmeshprovisioner V/MeshManagerApi: Provisioning pdu sent: 0x03000A
06-09 23:01:24.285 23694-23719/no.nordicsemi.android.nrfmeshprovisioner V/MeshManagerApi: Received provisioning message: 0x03010100010000040007000000
06-09 23:01:39.306 23694-23719/no.nordicsemi.android.nrfmeshprovisioner V/ProvisioningInvite: Number of elements: 1
06-09 23:01:39.310 23694-23719/no.nordicsemi.android.nrfmeshprovisioner V/ProvisioningInvite: Algorithm: 1
06-09 23:01:39.311 23694-23719/no.nordicsemi.android.nrfmeshprovisioner V/ProvisioningInvite: Public key type: Public key information unavailable
06-09 23:01:39.314 23694-23719/no.nordicsemi.android.nrfmeshprovisioner V/ProvisioningInvite: Static OOB type: Static OOB Actions unavailable
06-09 23:01:39.315 23694-23719/no.nordicsemi.android.nrfmeshprovisioner V/ProvisioningInvite: Output OOB size: 4
06-09 23:01:39.317 23694-23719/no.nordicsemi.android.nrfmeshprovisioner V/ProvisioningInvite: Output OOB action: Unknown
06-09 23:01:39.318 23694-23719/no.nordicsemi.android.nrfmeshprovisioner V/ProvisioningInvite: Input OOB size: 0
06-09 23:01:39.320 23694-23719/no.nordicsemi.android.nrfmeshprovisioner V/ProvisioningInvite: Input OOB action: No Input
06-09 23:01:39.323 23694-23719/no.nordicsemi.android.nrfmeshprovisioner V/ProvisioningStart: Provisioning start PDU: 0x03020000000004
06-09 23:01:39.327 23694-23719/no.nordicsemi.android.nrfmeshprovisioner V/BleMeshManager: Sending data : 0x03020000000004
06-09 23:01:39.338 23694-23720/no.nordicsemi.android.nrfmeshprovisioner V/BleMeshManager: Data written: 0x03020000000004
06-09 23:01:39.338 23694-23720/no.nordicsemi.android.nrfmeshprovisioner V/MeshManagerApi: Provisioning pdu sent: 0x03020000000004
06-09 23:01:39.525 23694-23701/no.nordicsemi.android.nrfmeshprovisioner I/zygote64: Compiler allocated 6MB to compile void android.view.ViewRootImpl.performTraversals()
06-09 23:01:41.104 23694-23720/no.nordicsemi.android.nrfmeshprovisioner V/ProvisioningPublicKey: X: length: 32 9C62CFE4201131671AB81D5D79AB17D18AA98ECFFC6E432BD507C15C76E2EB9D
06-09 23:01:41.105 23694-23720/no.nordicsemi.android.nrfmeshprovisioner V/ProvisioningPublicKey: Y: length: 32 500E2CF5658F56440CDD7C522C3545DB4C838C5C8D4C4F33DB00022D55105E31
    XY: 0x9C62CFE4201131671AB81D5D79AB17D18AA98ECFFC6E432BD507C15C76E2EB9D500E2CF5658F56440CDD7C522C3545DB4C838C5C8D4C4F33DB00022D55105E31
06-09 23:01:41.107 23694-23720/no.nordicsemi.android.nrfmeshprovisioner V/BleMeshManager: Sending data : 0x03039C62CFE4201131671AB81D5D79AB17D18AA98ECFFC6E432BD507C15C76E2EB9D500E2CF5658F56440CDD7C522C3545DB4C838C5C8D4C4F33DB00022D55105E31
06-09 23:01:41.113 23694-23720/no.nordicsemi.android.nrfmeshprovisioner V/MeshManagerApi: Received provisioning message: 0x030902
06-09 23:01:47.531 23694-23720/no.nordicsemi.android.nrfmeshprovisioner V/BleMeshManager: Data written: 0x03039C62CFE4201131671AB81D5D79AB17D18AA98ECFFC6E432BD507C15C76E2EB9D500E2CF5658F56440CDD7C522C3545DB4C838C5C8D4C4F33DB00022D55105E31
06-09 23:01:47.532 23694-23720/no.nordicsemi.android.nrfmeshprovisioner V/MeshManagerApi: Provisioning pdu sent: 0x03039C62CFE4201131671AB81D5D79AB17D18AA98ECFFC6E432BD507C15C76E2EB9D500E2CF5658F56440CDD7C522C3545DB4C838C5C8D4C4F33DB00022D55105E31
06-09 23:01:47.536 23694-23720/no.nordicsemi.android.nrfmeshprovisioner V/MeshManagerApi: Received provisioning message: 0x0303294E9F14E777304B9D7C2206DCCC9D6EF2B59B91EDB2D75FBFEE29A10439D585E83279CB0B0E72572CCEF58FB3EB078DC8250DBEDB5F8F0A6AC446F629A52709
06-09 23:02:13.026 23694-23694/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: cancelOpen() - device: 19:A4:5B:3C:6A:D0
06-09 23:02:13.062 23694-23738/no.nordicsemi.android.nrfmeshprovisioner D/OpenGLRenderer: endAllActiveAnimators on 0x7bf09e2800 (RippleDrawable) with handle 0x7bf09cbb40
06-09 23:02:13.113 23694-23694/no.nordicsemi.android.nrfmeshprovisioner V/ScannerFragment: scan started
06-09 23:02:13.125 23694-23694/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothAdapter: isLeEnabled(): ON
06-09 23:02:13.131 23694-23719/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=7 mScannerId=0
06-09 23:02:14.666 23694-23720/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=8 device=19:A4:5B:3C:6A:D0
06-09 23:02:14.675 23694-23720/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: close()
```

**PS:** Great thanks for the lib! You are the only ones (at least the only ones I've found) providing some libraries to communicate with ble meshes!
My intent is to implement PC library in JS to run on laptops or raspberry pis (or even Web Bluetooth), but I'm still struggling with the protocol, especially with provision part.

roshanrajaratnam, 2018-06-11:
@kabbi Thanks for reporting this issue, I will look in to this. About the message being displayed, there's definitely a typo involved there. Could you also confirm what type of OOB was used for this mesh node? And its great that you are taking the initiative to write something for desktop, it will be very useful for a lot of people out there.

roshanrajaratnam, 2018-06-13:
@kabbi Is there any chance I could get hold of the fw you are using for testing?

kabbi, 2018-06-14:
@roshanrajaratnam I'm using number-based OOB, my device outputs a number in the log that I'm to input into provisioning device.
I'm using mostly unmodified `blemesh` example app from [mynewt](https://mynewt.apache.org/latest/network/ble/ble_blemesh/) os. I can provide the binaries, but you will need their tool (`newt`) to upload them, as I don't think pure hex will be enough.

roshanrajaratnam, 2018-06-25:
Hi @kabbi, the latest version on the device branch has the fix related to provisioning failure. This was due to a bug in parsing output action values.

closed 2018-06-25 by roshanrajaratnam

kabbi, 2018-06-26:
Works perfectly, thank you!

PR #13 (2018-06-22): Bugfix/provisioning failure mynewt mesh example
fixes #2 out of bound action miscalculation causing the provisioning to fail