nordicsemi/Android-nRF-Mesh-Library #6: Error while building basic example

xaviergmrinq, 2018-06-20:
---

**Random values.xml pops up and build fails**
When I build the project, it fails with the message:

> ..\Android-nRF-Mesh-Library\Example\nrf-mesh\app\build\intermediates\res\merged\debug\values\values.xml
> Error:(1073) Multiple substitutions specified in non-positional format; did you mean to add the formatted="false" attribute?
> Error:(1073) Unexpected end tag string

Error is in line 1073 of this file. Tried many solutions given on the net and they don't work. Also opening of resources tag:
`<resources xmlns:ns1="urn:oasis:names:tc:xliff:document:1.2" xmlns:ns2="http://schemas.android.com/tools">`
The values of ns1= and ns2= values are red in colour i.e. error.
What I did here is removed these ns1 and ns2 values and converted it to:
`<resources>`
I then rebuilt it.

This time the error was different. It was in the file no\nordicsemi\android\meshprovisioner\transport\LowerTransportLayer.java
It gives an error for:
`mHandler.postDelayed(() -> {`
I says:

> Error:(476, 34) error: cannot find symbol method metafactory(Lookup,String,MethodType,MethodType,MethodHandle,MethodType)

**Platform details:**
 - Device: HP ZBook (8 GB RAM)
 - OS: Windows 10
 - Android 2.3.3
- Version of repository on 20 Jun 2018 at 17:30 IST

**Logs / Screenshots**
Below is my app gradle file:

```
apply plugin: 'com.android.application'

android {
    compileSdkVersion 27
    buildToolsVersion "27.0.3"

    defaultConfig {
        applicationId "no.nordicsemi.android.nrfmeshprovisioner"
        minSdkVersion 18
        targetSdkVersion 27
        versionCode 5
        versionName "0.2"
        testInstrumentationRunner "android.support.test.runner.AndroidJUnitRunner"
        multiDexEnabled true
        vectorDrawables.useSupportLibrary = true

        jackOptions {
            enabled true
        }
    }
    buildTypes {
        release {
            minifyEnabled false
            shrinkResources false
            useProguard true
            proguardFiles getDefaultProguardFile('proguard-android.txt'), 'proguard-rules.pro'
        }
    }

    compileOptions {
        targetCompatibility JavaVersion.VERSION_1_8
        sourceCompatibility JavaVersion.VERSION_1_8
    }

    dexOptions {
        preDexLibraries = false
    }
}
dependencies {
    compile fileTree(include: ['*.jar'], dir: 'libs')
    compile 'com.android.support:appcompat-v7:27.1.1'
    compile 'com.android.support:design:27.1.1'
    compile 'com.android.support:recyclerview-v7:27.1.1'
    compile 'com.android.support:cardview-v7:27.1.1'
    compile 'com.android.support.constraint:constraint-layout:1.1.0'
    // Butter Knife
    compile 'com.jakewharton:butterknife:8.8.1'
    annotationProcessor 'com.jakewharton:butterknife-compiler:8.8.1'
    // Brings the new BluetoothLeScanner API to older platforms
    compile 'no.nordicsemi.android.support.v18:scanner:1.1.0'
    compile 'no.nordicsemi.android:ble:1.2.0'
    // Log Bluetooth LE events in nRF Logger
    compile 'com.android.support:multidex:1.0.1'
    // Lifecycle extensions
    compile 'android.arch.lifecycle:extensions:1.1.0'
    compile 'com.google.dagger:dagger:2.11'
    compile 'com.google.dagger:dagger-android:2.11'
    compile 'com.google.dagger:dagger-android-support:2.11'
    annotationProcessor 'com.google.dagger:dagger-android-processor:2.11'
    annotationProcessor 'com.google.dagger:dagger-compiler:2.11'
    compile 'com.google.code.gson:gson:2.8.3'
    compile project(':meshprovisioner')
}

```

roshanrajaratnam, 2018-06-21:
@xaviergmrinq sorry for any inconvenience caused. This is quite strange have you tried a clean build?

xaviergmrinq, 2018-06-21:
@roshanrajaratnam Yes I have tried a clean build

roshanrajaratnam, 2018-06-21:
@xaviergmrinq Have you cloned the project as is and built it or have you made any changes in the settings.gradle file? If not could you try one more thing by invalidating cache on Android Studio to see if its of any help?

xaviergmrinq, 2018-06-21:
@roshanrajaratnam Thanks a lot for your help. I had actually changed the gradel keyword 'implementation' in your code to 'compile' because it was not supported for my 2.3.3 version of Android Studio. After seeing your comment it dawned on me that I should go and update my Android IDE. I updated my version to 3.1.3. Changed back the gradel keywords to 'implementation' and it worked.

closed 2018-06-21 by xaviergmrinq