nordicsemi/Android-nRF-Mesh-Library #9: Switch the light on Android

kcjian, 2018-06-21:
Hi,
Can you tell me how to use an Android application to switch lights, such as the iOS application nRF Mesh?
Or can you add this feature to the Android application nRF Mesh?
Thank you!

xaviergmrinq, 2018-06-21:
I had a similar doubt but it's more specific and not general. But I decided against creating a separate issue and I'm appending my query here.
@roshanrajaratnam I would like to take forward what @kcjian is trying to say in more technical terms (and @kcjian comment if what I'm asking for doesn't match with what you want).

@roshanrajaratnam Correct me if I go wrong on this.
I was going through the code and trying to figure out how to put data into the Proxy node i.e. PDU.
3 of the files caught my interest:

1. `meshprovisioner/MeshManagerApi.java`                     (as in meshprovisioner module)
2. `nrfmeshprovisioner/ble/BleMeshManager.java`           (as in meshprovisioner module)
3. `ble/BleManager.java`                                                     (as in ble module)

From these I feel 1 is related to Provisioning of the mesh and 2, 3 are related to sending data to the lower level Android bluetooth manager.
I've used the Bluetooth SIG App Accelerator library/files wherein they give a BleWrapper object and writing data to/reading data from a Bluetooth device works by writing/reading to a **Characteristic** of a **Service** of a **GATT server** running on a Bluetooth device. Now that's how it is for Bluetooth devices which are not Mesh and one to one Bluetooth communication.
I could see in file 2 above the PROXY_IN and PROXY_OUT characteristic mentioned which makes sense when compared with the **Bluetooth SIG Mesh Model** document.

What confuses me is that this project needs both the 'ble' and 'meshprovisioner' module. So, there is both the `BleMeshManager.java` and `BleManager.java`. Won't their working collide with each other?

But, that's not my main question in this thread is. My main interest is to write to/read from proxy characteristics of the mesh and that will solve what @kcjian is looking for.

**I can see in `BleMeshManager.java` that there is a function `sendPdu(final byte[] pdu)`.
Is this the function I have to call from my MainActivity like on button press in a graphical app?**

If not then can you give a code snippet of which function in which file I've to call from my MainActivity to send data PDUs (byte[] array) to mesh nodes?

Is it something like this: `instanceBleMeshManager.sendPdu((byte[]) {0x01, 0x02, 0x03})` I have to call from the MainActivity?
If not then give a code snippet.
Also give a code snippet of how to read from a PROXY_OUT characteristic of the Bluetooth service.

Thanks in advance

roshanrajaratnam, 2018-06-21:
@kcjian I have implemented controlling generic light server model which is to turn on lights using tthe application. This feature is not pushed to play store yet but it will be by tomorrow. If you clone the latest [dev branch](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/tree/dev) you should be able to try this feature. Let me know how it goes.

@xaviergmrinq I would be really grateful if you could open a new issue related to your query.

xaviergmrinq, 2018-06-22:
In which activity of Dev branch is this controlling part there? It would be even better if you tell me the part of the code which does it.

roshanrajaratnam, 2018-06-22:
@xaviergmrinq ModelConfigurationActivity contains the controls to turn on/off light with a Generic OnOff Set Unacknowledged message. Please feel free to close the issue if this helps.

kcjian, 2018-06-22:
@roshanrajaratnam   I find it,thanks for your help。And do you know where there is detailed documentation about the process of generating pdu, because we don't just want to turn on the lights，but just looking at the code is hard to understand.

closed 2018-06-22 by kcjian

xaviergmrinq, 2018-06-22:
@kcjian is right. I too want to send raw byte PDUs. Instead of `MeshManagerApi.setGenericOnOff()`.
`MeshManagerApi.sendPdu(BaseMeshNode meshNode, byte[] pdu);` looks correct to me. But what is BaseMeshNode? what should I enter here while calling `sendPdu()` function? Is it a byte, Hex, Int .etc.?

Or is it that I can use `instanceBleMeshManager.sendPdu(byte[] pdu)` instead of it.

kcjian, 2018-06-22:
@xaviergmrinq 

 hi，

1. I think you can look this [https://mp.weixin.qq.com/s?__biz=MzA4MTg3MTg2Mg==&mid=2651236886&idx=2&sn=12a439d75fffd15d2e837ec339e5cbb2&chksm=847c5c8fb30bd5996515ce0f35f85645c2e1952b3d519a8a51c0788009ce1ceee8892f92c5ab&scene=21#wechat_redirect](url) to understand the basic concepts.

2. sendPdu() need a byte array

3. The result of MeshManagerApi.sendPdu() is using BleMeshManager.sendPdu() via MeshService.sendPdu()。

xaviergmrinq, 2018-06-22:
Hi @kcjian ,
The link you shared in point 1 doesn't work.