nordicsemi/Android-nRF-Mesh-Library #10: Sending data PDU byte array to the mesh through GUI on app

xaviergmrinq, 2018-06-22:
@roshanrajaratnam Correct me if I go wrong on this.
I was going through the code and trying to figure out how to put data into the Proxy node i.e. PDU.
3 of the files caught my interest:

1. `meshprovisioner/MeshManagerApi.java`                     (as in meshprovisioner module)
2. `nrfmeshprovisioner/ble/BleMeshManager.java`           (as in meshprovisioner module)
3. `ble/BleManager.java`                                                     (as in ble module)

From these I feel 1 is related to Provisioning of the mesh and 2, 3 are related to sending data to the lower level Android Bluetooth manager.
I've used the Bluetooth SIG App Accelerator library/files wherein they give a BleWrapper object and writing data to/reading data from a Bluetooth device works by writing/reading to a **Characteristic** of a **Service** of a **GATT server** running on a Bluetooth device. Now that's how it is for Bluetooth devices which are not Mesh and one to one Bluetooth communication.
I could see in file 2 above the PROXY_IN and PROXY_OUT characteristic mentioned which makes sense when compared with the **Bluetooth SIG Mesh Model** document.

What confuses me is that this project needs both the 'ble' and 'meshprovisioner' module. So, there is both the `BleMeshManager.java` and `BleManager.java`. Won't their working collide with each other?

But, that's not my main question in this thread is. My main interest is to write to/read from proxy characteristics of the mesh and that will solve what @kcjian is looking for.

**I can see in `BleMeshManager.java` that there is a function `sendPdu(final byte[] pdu)`.
Is this the function I have to call from my MainActivity like on button press in a graphical app?**

If not then can you give a code snippet of which function in which file I've to call from my MainActivity to send data PDUs (byte[] array) to mesh nodes?

Is it something like this: `instanceBleMeshManager.sendPdu((byte[]) {0x01, 0x02, 0x03})` I have to call from the MainActivity?
If not then give a code snippet.
Also, give a code snippet of how to read from a PROXY_OUT characteristic of the Bluetooth service.

Thanks in advance

roshanrajaratnam, 2018-06-22:
@xaviergmrinq adding to the points you made above

1. MeshManagerApi is the public api for the mesh provisioner module.
2. BleMeshManager extends the BleManager, this way we can override certain methods from the BleManager to use in our BleManager. So you are correct that point 2,3 are related to ble gatt level stuff. 

If you have used our nRF Toolbox examples before you may see the same design where users of the library can override certain methods to suit their needs. The idea was to keep the implementation as similar as possible so that it's much easier for existing users and to avoid introducing new designs to different apps.

We could have added our android ble library in to the meshprovisioner library but we decided against it so that users have the freedom to use their own ble libraries with the meshprovisioner library. Ideally I would recommend to use our own BleManager library and that would make things much easier to debug if you encounter any issues.

In order to send a mesh pdu all you have to do is call the related methods to from the MeshManagerApi. for example if you want to turn on/off a GenericOnOffServerModel all you have to do is just to call MeshManagerApi.setGenericOnOff() and the application will handle sending the pdu and parsing received pdus by itself.

xaviergmrinq, 2018-06-22:
Carrying forward from @kcjian 's thread because I felt that it being closed @roshanrajaratnam may overlook it.
I too want to send raw byte PDUs. How to send it with or without source, destination address, you tell me how to do it.
Instead of `MeshManagerApi.setGenericOnOff()`. `MeshManagerApi.sendPdu(BaseMeshNode meshNode, byte[] pdu);` looks correct to me. But what is BaseMeshNode? what should I enter here while calling `sendPdu()` function? Is it a byte, Hex, Int .etc.?

Or is it that I can use `instanceBleMeshManager.sendPdu(byte[] pdu)` instead of it.

xaviergmrinq, 2018-06-26:
Ok, @roshanrajaratnam or @kcjian . I need something that I can call on the press of a button in an activity. I need some documentation on how to send data from my UI activity and receive data from the mesh in the UI activity. I don't know what BaseMeshNode is and how to initialix\ze it in my activity. Please point me to some good documentation to send such PDUs from my UI activity.

roshanrajaratnam, 2018-06-26:
@xaviergmrinq what are you trying to achieve? What do you want to send as raw bytes, is it a particular mesh pdu? If you look at the MeshService.java you will see that you can call mBleMeshManager.sendPdu(bytearray) and send whatever data you want. You cannot initialize BaseMeshNode since its an abstract class. I'm not entirely sure how you're app's design is so you will have to create your own method to call sendPdu from your activity.

xaviergmrinq, 2018-07-02:
Hi, @roshanrajaratnam I wanted to try out a code where I have my own activity and I can jump to your MainActivity.java based on a button click. But, When I do so I get an exception saying that there is no Injector factory bound with the class. I did a search online and mostly got resources on Dagger but I guess this project used ButterKnife. I'm new to Bind, Injector and ButterKnife and don't know how to get around this UI impediment. Can you point me to some resource which can solve this problem or can you help me on it?

xaviergmrinq, 2018-07-04:
@roshanrajaratnam Also, [](https://devzone.nordicsemi.com/members/edvin-holmseth) told me that you are already looking into it.
Have a look at my problem in detail below:
[](https://devzone.nordicsemi.com/f/nordic-q-a/36029/unable-to-call-the-mainactivity-of-android-nrf-mesh-library-from-my-activity-in-android-for-my-version-of-nrf-mesh-provisioner-app)