nordicsemi/Android-nRF-Mesh-Library #20: Provisioning the second Thingy will failed

kcjian, 2018-07-03:
---

**Describe the bug**
Provisioning the second Thingy will failed，happen to Example/nrf-mesh

**To Reproduce**
Steps to reproduce the behavior:
1. Provisioning a Thingy,Succeed
2. Provisioning the second Thingy , Failed always

roshanrajaratnam, 2018-07-11:
Hi, sorry for the inconvenience, I just noticed this issue and I'll look in to it soon. what's  the fw you have used on the Thingy? Any chance you could provide it, if its a custom fw?

roshanrajaratnam, 2018-07-19:
@kcjian any updates on this issue? Was there some kind of an error that was thrown while provisioning the second thingy? What fw did you use for this? If its a custom fw any chance you could share it with me?

kcjian, 2018-07-24:
Thank you for your reply, I thought i solved this problem by modifying the file MeshService.java
[image]

closed 2018-07-24 by kcjian

roshanrajaratnam, 2018-07-24:
Great thanks!