nordicsemi/Android-nRF-Mesh-Library #25: Vendor Model

kcjian, 2018-07-25:
We hope to add a Vendor Model to config beacon,We consulted the GenericOnOffModel and made some necessary changes，implemented BeaconGet, BeaconSet,BeaconStatus；
we can receive BeaconStatus when i get Beacon Data,No matter how long the data is.
we can receive BeaconStatus when I send a short data to set beacon ,that valid data about 6 bytes ,example beacon major.
But we can not receive any data when I send a long data to set beacon ,that valid data about 20 bytes ,example beacon uuid，My colleague said he didn't receive the data of setting.
The above data length include our protocol but does not include mesh protocol.
Can you give me any advice，thanks！

kcjian, 2018-07-25:
Additional information , App can receive long bytes data but can not send ,It seems that  the data is discarded by the hardware  because of the length

kcjian, 2018-07-25:
oh,It looks like I've solved it, Do a little change in the library  [LowerTransportLayer.java](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/master/android-nrf-mesh-library/meshprovisioner/src/main/java/no/nordicsemi/android/meshprovisioner/transport/LowerTransportLayer.java)，Can you help me see it，thanks！
[image]

roshanrajaratnam, 2018-07-25:
@kcjian This is a bit strange that this is not working for you. This should work out of the box in my opinion. If you notice the AppKeyAdd message which is a segmented access message containing exactly 20 bytes and the best way for you to test is by adding an app key to you vendor model.

This change does not look right to me since you are assigning your length only if your segO is at a particular segment index. The current implementation handles it irrespective of a particular segO. I also feel that this might work only if your message  has 2 segments and might break other situations with messages that may contain more than 2 segments.

kcjian, 2018-07-25:
@roshanrajaratnam  Yes,if the message only has 1 segments ,the code won't go here,even if code go here ,this "length" not important. I just want they are work when the message has 2 segments or more.The test found the same as I thought, included adding an app key to vendor model.

kcjian, 2018-07-25:
My change is because I found when the message has 3 segments and when the seg0=1,the "length" is bigger than “MAX_SEGMENTED_ACCESS_PAYLOAD_LENGTH”.

roshanrajaratnam, 2018-07-25:
@kcjian I'm not sure I understand you correctly here. Did app key add work for you without the changes you made on LowerTransportLayer? It should work for you because I have tested that and it seemed to be working fine.

I will test with some garbage values more than 20 bytes to create more than 2 segments to see how it behaves. I'll get back to you on this.

roshanrajaratnam, 2018-07-27:
@kcjian thanks for reporting this issue, you were correct about the segmentation not working properly when the number of segments increased more than 2. The fix was to calculate the length of the segmented payload length (excluding the header) dynamically. I will close this issue since I have tested this internally and seemed to be working fine. Let me know if this works for you.

closed 2018-07-27 by roshanrajaratnam