nordicsemi/Android-nRF-Mesh-Library #30: Mesh

kcjian, 2018-07-31:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

roshanrajaratnam, 2018-08-06:
@kcjian Hi this is something we will be looking in to pretty soon.

zanibonilighting, 2019-01-24:
hi @roshanrajaratnam may I ask if there is any update on this one?

roshanrajaratnam, 2019-01-25:
@zanibonilighting Currently there is no update on this one right now unfortunately!

NomanAziz51, 2022-05-09:
Hi, @roshanrajaratnam, Is there any update on this module?

connyhald, 2024-02-12:
Reading this nordic page here: https://developer.nordicsemi.com/nRF_Connect_SDK/doc/2.4.0/nrf/samples/bluetooth/mesh/dfu/distributor/README.html

It sound like on the app side we would need:
- `Firmware Distribution Client `
- `Firmware Update Client`
- `BLOB Transfer Client`

Those models are currently not available as part of this lib here, right?

roshanrajaratnam, 2026-08-27:
Unfortunately we don't plan to add Mesh DFU in to library. We are currently working on adding DFU support to the new [Kotlin Mesh Library](https://github.com/nordicsemi/Kotlin-Mesh-Library)

closed 2026-08-27 by roshanrajaratnam