nordicsemi/Android-nRF-Mesh-Library #61: Select mesh node based on src address

roshanrajaratnam, 2018-09-18:
Currently the app disregards the src of the messages and assumes that its the same node as the user is already communicating with. This may lead to wrong node being updated when receiving messages.

DomiR, 2018-11-16:
This is very important to us, do you have an ETA? We need to get updates from a node, based on the publish group it is sending messages to, as we want to be able to control the light via external button and still get the update in our app. BUTTON (should publish to 0xC001)=>(will subscribe to 0xC001) GENRIC ON OFF (should publish to 0xC002) => App (get on off changes)

roshanrajaratnam, 2018-11-19:
@DomiR As a matter of fact this is already in place in the library. I might have missed out on referencing this issue. If configured to suit your needs, app will get on/off commands in addition to the status (if you are sending acknowledged on/off). IMHO you should have the light bulb publish the status back to the app instead of having the app listen to the on/off message as well as the status on the subscribe group.

Still to get continuous updates you will have to implement the Proxy Whitelist message to receive continuous status updates. You have to send a message to the proxy to get one back unless your app's unicast is on the whitelist of the proxy. Implementing the proxy whitelist message would fix this for you. This is something I plan on adding, i'm currently working on importing and exporting networks so i would say, probably by end of next week I should have the whitelisting in place.

PR #127 (2019-01-16): Preparing for release v1.1.0
* Brand new Nordic Identity colors!!!!
* Fixes #73
  - Added message classes and removed states for status messages.
* Fixes #32 
  - Added server configuration messages such as NetworkTransmit, ProxyState and Relay.
* Fixes #36 
  - Supports importing and exporting mesh networks. This allows users to share network across nRF Mesh Apps.
  - Added RoomDB to mesh library replacing shared preference storage. 
* Fixes #40 
  - Library supports deleting nodes.
* Fixes #61 
  - Selects a node based on the node address which includes element address. In case of a certain  [... cut]
closed 2019-01-16 by roshanrajaratnam