nordicsemi/Android-nRF-Mesh-Library #65: We would like to provision multiple networks

DomiR, 2018-09-20:
**Is your feature request related to a problem? Please describe.**
The newest version of this library saves the provisioning settings to SharedPreferences, but should probably be passed to the MeshApi in some way.

**Describe the solution you'd like**
Pass the networkkey and application keys to the api like in the previous version, which would allow to manage multiple networks.

What's the strategy on this? Reading them from SharedPreferences does not seem to be an optimal solution for a library and should rather be taken care of by the application.

DomiR, 2018-09-21:
@roshanrajaratnam thank you very much for your clarifications. We would like to offer support for multiple networks in our app and would appreciate a solution for this as well.

In the previous version we were able to 

I would like to propose more

roshanrajaratnam, 2018-09-21:
@DomiR I just seen your edited comment, thanks for the suggestions. I will take a look in to this as I get time.

roshanrajaratnam, 2018-12-05:
@DomiR #101 have added the capabilities for this by migrating the existing preferences to the DB,. Currently the api nor the app supports this feature but the DB is designed to handle this. Let me know your thoughts about this!

DomiR, 2018-12-12:
Hey @roshanrajaratnam, I’ve been testing the new branch for a while now and it seems to be working fine, thank you very much for making the export work. I would like to extend the api to allow exporting the network to a json string directly, as we would like to send it via email and we don’t want to add file access permissions just to temporarily save files.

Another thing I’ve been noticing, is that the sequence number is not stored to the db but is reset every time the app is restarted. I will add a PR for this a bit later today.

roshanrajaratnam, 2018-12-12:
@DomiR Sure send me a PR once you extend it to work towards the JSON string.
I had seen this bug, but it was fixed, if not the messages should not continue to work upon the app restart.

closed 2019-01-16 by roshanrajaratnam