nordicsemi/Android-nRF-Mesh-Library #72: Add support for sending real time Generic Level values

roshanrajaratnam, 2018-09-28:
As a user/developer this would allow setting real time values to a generic level mode.

closed 2019-03-28 by roshanrajaratnam

DevelopertwoAtBubblyNet, 2019-03-28:
May I ask why this was originally added, then removed, and then eventually closed? This is a feature I have been attempting to get working but have been struggling with. Any hints on how to go about adding it? I understand its probably not a top priority for you.

roshanrajaratnam, 2019-03-29:
Hi @DevelopertwoAtBubblyNet, this is more like a feature towards the app not the library itself so its not a top priority for us, but it still remains in our internal backlog. One way to implement this is to divide the seekbar in to 10 or 20 points and when reaching each point you can send an unacknowledged generic level message, this way you will not overwhelm with so many messages.