nordicsemi/Android-nRF-Mesh-Library #75: Handle multiple generic onoff server models with different elements on same device

DomiR, 2018-10-10:
**Is your feature request related to a problem? Please describe.**
We have devices with multiple elements, that all support genericOnOffServer. Sending acknowledged messages return status messages, but it is not clear, which element sent which message. Currently the callbacks for onGenericOnOffStatusReceived won't allow to determine from which element these messages came from.

`public void onGenericOnOffStatusReceived(ProvisionedMeshNode node, boolean presentOnOff, Boolean targetOnOff, int transitionSteps, int transitionResolution)`

**Describe the solution you'd like**
We would like to have `public void onGenericOnOffStatusReceived(ProvisionedMeshNode node, Integer elementAddress, boolean presentOnOff, Boolean targetOnOff, int transitionSteps, int transitionResolution)` where **Integer elementAddress** (or if preferred byte[] elementAddress) is added to all callbacks, that are coming from a specific element, but does not have the element address in the actual payload.

I'd be happy to submit a pull request. Is there any eta on #73 so I can wait for it?

roshanrajaratnam, 2018-10-11:
@DomiR Yes this was something missing in the library and is being worked on as we speak. This issue will be fixed once I'm done implementing #73. The plan is that I should have it ready by next week.

DomiR, 2018-10-11:
Hey @roshanrajaratnam, thank you for your reply. I saw that you are restructuring callbacks. Adding access to the message or the src of the message would be nice. If I can help, let me know 👍

roshanrajaratnam, 2018-10-23:
@DomiR Sorry for the delay on thsi and I have been away from work for a few days. 

I have made a quite a lot of changes to the internals of the library. I also tried to keep the breaking changes to a minimum on the API. Some feedback would be really appreciated, on [this](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/tree/improvements/fix-repository) branch. As an example getting composition data from a node will look like the following sendConfigMessMessage(ConfigCompositionDataGet)`

DomiR, 2018-10-25:
Wow, this was a lot of work, thank you very much for this huge effort. I've been checking out the new structure and the new api feels much cleaner as one only needs to implement one callback and handle just the necessary cases. I have a bit of rewriting to do to test this in our app, but from what I can tell these are much welcomed improvements.

roshanrajaratnam, 2018-10-26:
@DomiR Thanks for the feedback. Some of these changes were much needed and had to be addressed at some point. Sooner the better I suppose. Anyways hope there's not too many breaking changes.

PR #83 (2018-10-26): Improvements/add builder for mesh messages #73
* Fixed #73 by adding wrapper classes mesh messages.
* Adds a generic callbacks fore received/sent mesh messages. This has also removed individual callback per mesh message sent/received based on the mesh message type be it configuration/application messages.
* Fixed #75 by adding source address of the message to callbacks to notify where the message originated from.
* Decouple unprovisioned and provisioned mesh node classes, this also improves the scoping of the base classes.
* Fixes a bug introduced while refactoring GenericLevelGetState.
closed 2019-01-16 by roshanrajaratnam