nordicsemi/Android-nRF-Mesh-Library #81: ConfigRelaySet

WangManAndroid, 2018-10-25:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

roshanrajaratnam, 2018-10-25:
@WangManAndroid Thanks for sending this feature request, but don't close after openning one unless you don't want it ;) I'll re-open it again. Duplicate of #32

WangManAndroid, 2018-10-25:
hao to make a file for ConfigRelaySet  function?

kcjian, 2018-10-30:
`package no.nordicsemi.android.meshprovisioner.configuration;

import android.content.Context;
import android.util.Log;

import no.nordicsemi.android.meshprovisioner.InternalTransportCallbacks;
import no.nordicsemi.android.meshprovisioner.MeshConfigurationStatusCallbacks;
import no.nordicsemi.android.meshprovisioner.messages.AccessMessage;
import no.nordicsemi.android.meshprovisioner.messages.ControlMessage;
import no.nordicsemi.android.meshprovisioner.opcodes.ConfigMessageOpCodes;
import no.nordicsemi.android.meshprovisioner.utils.MeshParserUtils;

public class ConfigRelaySet extends ConfigMessage {

    private static final String TAG = ConfigRelaySet.class.getSimpleName();

    private final int mAszmic;
    private int relay,relayTransmit;

    public ConfigRelaySet(final Context context, final ProvisionedMeshNode provisionedMeshNode, final boolean aszmic,
                          final InternalTransportCallbacks mInternalTransportCallbacks,
                          final MeshConfigurationStatusCallbacks meshConfigurationStatusCallbacks, int relay, int relayTransmit)  {
        super(context, provisionedMeshNode);
        this.mAszmic = aszmic ? 1 : 0;
        this.mInternalTransportCallbacks = mInternalTransportCallbacks;
        this.mConfigStatusCallbacks = meshConfigurationStatusCallbacks;
        this.relay=relay;
        this.relayTransmit=relayTransmit;
        createAccessMessage();
    }

    @Override
    public MessageState getState() {
        return MessageState.CONFIG_RELAY_SET;
    }

    /**
     * Creates the access message to be sent to the node
     */
    private void createAccessMessage() {
        final byte[] key = mProvisionedMeshNode.getDeviceKey();
        int akf = 0;
        int aid = 0;
        byte[] parameters=new byte[2];
        parameters[0]= (byte) relay;
        parameters[1]= (byte) (byte) (relayTransmit);

        final AccessMessage accessMessage = mMeshTransport.createMeshMessage(mProvisionedMeshNode, mSrc, key, akf, aid, mAszmic,
                ConfigMessageOpCodes.CONFIG_RELAY_SET, parameters);
        mPayloads.putAll(accessMessage.getNetworkPdu());
    }

    /**
     * Starts sending the mesh pdu
     */
    public void executeSend() {
        if (!mPayloads.isEmpty()) {
            for (int i = 0; i < mPayloads.size(); i++) {
                mInternalTransportCallbacks.sendPdu(mProvisionedMeshNode, mPayloads.get(i));
            }

//            if (mConfigStatusCallbacks != null)
//                mConfigStatusCallbacks.onMeshNodeResetSent(mProvisionedMeshNode);
        }
    }

    @Override
    public void sendSegmentAcknowledgementMessage(final ControlMessage controlMessage) {
        final ControlMessage message = mMeshTransport.createSegmentBlockAcknowledgementMessage(controlMessage);
        Log.v(TAG, "Sending acknowledgement: " + MeshParserUtils.bytesToHex(message.getNetworkPdu().get(0), false));
        mInternalTransportCallbacks.sendPdu(mProvisionedMeshNode, message.getNetworkPdu().get(0));
        mConfigStatusCallbacks.onBlockAcknowledgementSent(mProvisionedMeshNode);
    }
}
`

roshanrajaratnam, 2018-10-30:
@kcjian Thanks for this :+1:

 I would really appreciate if you could send this as pull request. This way I can make release it as a part of the library. 
Thanks again!

roshanrajaratnam, 2018-12-12:
Closing in favor of #32

closed 2018-12-12 by roshanrajaratnam