nordicsemi/Android-nRF-Mesh-Library #94: the bug of MeshModel

WangManAndroid, 2018-11-06:
---

`   protected MeshModel(final Parcel in) {

        final int modelId = in.readInt();
        if (modelId < Short.MIN_VALUE || modelId > Short.MAX_VALUE) {
            mModelId = modelId;
        } else {
            mModelId = (short) modelId;
        }
        in.readList(mBoundAppKeyIndexes, Integer.class.getClassLoader());
        sortAppKeys(in.readHashMap(String.class.getClassLoader()));
        publishAddress = in.createByteArray();
        publishAppKeyIndex = in.createByteArray();
        credentialFlag = in.readInt();
        publishTtl = in.readInt();
        publishPeriod = in.readInt();
//        publicationSteps =  publishPeriod >> 6;
//        publicationResolution = publishPeriod & 0x03;
        publicationSteps = (short) (publishPeriod >> 2);
        publicationResolution = publishPeriod & 0x03;
        publishRetransmitCount=in.readInt();//add it 2018/11/6
        publishRetransmitIntervalSteps =  in.readInt();
        in.readList(mSubscriptionAddress, byte[].class.getClassLoader());
    }`

WangManAndroid, 2018-11-06:
maby  you forget add it   >>  publishRetransmitCount=in.readInt();
                                          >>   dest.writeInt(publishRetransmitCount);

roshanrajaratnam, 2018-11-06:
@WangManAndroid Currently this has been moved in to a seperate class called PublicationSettings

closed 2018-11-06 by roshanrajaratnam