nordicsemi/Android-nRF-Mesh-Library #96: Tid

shengMR, 2018-11-12:
Hello, ask, I want to know where to modify the value of tid, because the value of tid is the same when sending data, but I can not find the attribute tid

roshanrajaratnam, 2018-11-12:
Hi, @shengMR as of now i'm using the sequence number as the TID, since its unique for every message. Hope this helps

closed 2018-11-12 by roshanrajaratnam

shengMR, 2018-11-12:
Well, understand, but why do I call the Api without return, it takes 6 seconds to send the next data? Can I have any way to change this phenomenon?

shengMR, 2018-11-13:
Excuse me, I still want to ask a question, that is, the data tid that I send through setGenericOnOffUnacknowledged will not change, but the data tid sent by setGenericOnOff will change.

roshanrajaratnam, 2018-11-13:
Hi both messages use the sequence number of that particular message, so it is unique. From what I see in the source they are unique, but I'll look in to this. Have you made any changes in there?

AISkynet, 2018-11-13:
> Hi both messages use the sequence number of that particular message, so it is unique. From what I see in the source they are unique, but I'll look in to this. Have you made any changes in there?

Hi, can you use the App to send the setGenericOnOffUnacknowledged cmd to control the led fluency from nrf52 development kit which flashed hex from the nrf5_SDK_for_Mesh_v2.2.0 -> light_switch -> proxy_server? I found out the tid is always the same one by the embedded IDE debug window.

shengMR, 2018-11-13:
Thank you, I found the SN, thank you for your help.

shengMR, 2018-11-13:
@roshanrajaratnam Thank you, I found the SN, thank you for your help.

roshanrajaratnam, 2018-11-13:
@shengMR sorry I just saw this. Hope everything is working as expected!