nordicsemi/Android-nRF-Mesh-Library #99: Publish Address 0xFFFF

shengMR, 2018-11-19:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

shengMR, 2018-11-19:
Ask, when I set the device's publishing address to 0xFFFF, this time I go to press the button on the development board, why the phone can not be synchronized?

roshanrajaratnam, 2018-11-19:
Hi, I'm not entirely sure what you are trying to achieve. Could you explain the setup a little bit more?

shengMR, 2018-11-19:
This is the case, I want to press the button to control the indicator light on the Nordic development board, then my phone can synchronize the switch state, but I used the release address set to 0xFFFF no effect, 0xFFFF not the entire network can receive it?

roshanrajaratnam, 2018-11-19:
Are you trying receive GenericOnOffSet or GenericOnOffStatus on the phone? If you want the GenericOnOffSet,  I would suggest to get the light bulb to publish the status to the phone instead. So when you're inside the model int he app you should get the status.

shengMR, 2018-11-19:
Well, this is the case, I really want to receive GenericOnOffStatus, but the source address of the phone is 0x07FF, I use the phone to set the Bluetooth module's publishing address to 0x07FF, this time press the button on the development board, my phone can be obtained through GenericOnOffStatus Switch state, this is no problem, the problem is set to 0xFFFF, it is not possible, is it wrong?

roshanrajaratnam, 2019-02-06:
Sorry for the delayed reply, you can now enable the blacklist filter on the proxy server you are connected to and then test this feature.

closed 2019-03-05 by roshanrajaratnam

taaltechIOT, 2020-06-24:
we have node 1 as generic on/off client with publish address 0xFFFF and then node 2,3 &4 as generic on/off server with subscribe address 0xC001. Doe this work? i assume being publish adress 0xFFFF which is broadcast, node 2,3 & 4 can receive the data even though the subscription address is different.

Can you please confirm

philips77, 2020-06-25:
Yes, that's true. 0xFFFF should reach all elements on all nodes in the same subnetwork.