nordicsemi/Android-nRF-Mesh-Library #103: Get BluetoothLeScannerCompat  from different  Android Version

WangManAndroid, 2018-12-05:
`
public static BluetoothLeScannerCompat getScanner() {
		if (mInstance != null)
			return mInstance;
		if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.O)
			return mInstance = new BluetoothLeScannerImplOreo();
		if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.M)
			return mInstance = new BluetoothLeScannerImplMarshmallow();
		if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.LOLLIPOP)
			return mInstance = new BluetoothLeScannerImplLollipop();
		return mInstance = new BluetoothLeScannerImplJB();
	}
`
in my   xiaomi 4>>will   get   BluetoothLeScannerImplMarshmallow (android 6.0)
               xiaomi mix 2>> will  get BluetoothLeScannerImplOreo( android 8.0)

 could you tell me the diffrence  of this?  
  i found  xiaomi mix2  can scan  provinsioned   device  easier  than xiaomi 4, 
 You can't even scan it on xiammi 4,
 i  don`t   know  the reaseon  of  this condition ,
 whiling  provinsioning , onProvisioningComplete,
` public void onProvisioningComplete(final ProvisionedMeshNode provisionedMeshNode) {
        provisionedMeshNode.setIsProvisioned(true);
        Debuger.d("onProvisioningComplete............");
        mMeshNode = provisionedMeshNode;
        mIsProvisioningComplete = true;

        final Intent intent = new Intent(ACTION_PROVISIONING_STATE);
        intent.putExtra(EXTRA_PROVISIONING_STATE, MeshNodeStates.MeshNodeStatus.PROVISIONING_COMPLETE.getState());
        LocalBroadcastManager.getInstance(this).sendBroadcast(intent);

        mIsProvisioningComplete = true;
        mIsReconnecting = true;
        final Intent intent1 = new Intent(ACTION_IS_RECONNECTING);
        intent1.putExtra(EXTRA_DATA, mIsReconnecting);
        LocalBroadcastManager.getInstance(this).sendBroadcast(intent1);

        //mIsReconnecting.postValue(true);
        mBleMeshManager.setProvisioningComplete(true);
        mBleMeshManager.disconnect();
        mBleMeshManager.refreshDeviceCache();
        mHandler.postDelayed(mReconnectRunnable, 1500); //Added a slight delay to disconnect and refresh the cache
    }`
in this place  the xiaomi 4  rescan  so hard  and  you can't even scan it on xiammi 4,
Is this a compatibility issue?

WangManAndroid, 2018-12-05:
my english  is poor  , idon`t know if  you can  know  the description of the i

WangManAndroid, 2018-12-05:
When the interface was stuck on connecting connecting lines, I found that the log was print  as follows:
onClientRegistered() - status=0 clientIf=6
onClientConnectionState() - status=133 clientIf=6 device=98:12:88:77:00:0B

roshanrajaratnam, 2018-12-05:
Hi @WangManAndroid different versions of Android have different Ble scanners. The Bluetooth scanner compat library is a backward compatible library that will give you a scanner instance based on the Android version.

I recommend you try the latest dev branch. Error 133 is a connection timeout. Also newer phones have always been more robust when it comes to reconnecting after provisioning, specially in Android Oreo onwards because it's very rarely you can see error 133.
Still there was a bug that was stopping the UI navigating back to the scanning fragment. So it looks like the app was stuck but ideally it found the device and failed to connect and it should have gone back to the scanner fragment. Could you try the latest version in the Dev branch and let me know if you face this?

WangManAndroid, 2018-12-05:
ok  i`ll   try  it ,  Thanks for your reply

WangManAndroid, 2018-12-10:
today   I found  new problems   while  Discovering services  ` gatt.discoverServices(); 
 @Override
        public void onConnectionStateChange(BluetoothGatt gatt, int status, int newState) {
          //  It's possible here    callback>  status=133
} `
in  some   android phone  will heppen     For example, some of xiaomi's mobile phones

roshanrajaratnam, 2018-12-10:
Yeah this is possible, you just have to work around it in my opinion. Sometimes this depends on the Android version on certain devices. You can also try turning on/off the bluetooth on the device to see if it helps?

roshanrajaratnam, 2018-12-10:
Well the only way around this is to make sure all the bluetooth related communication is done in the UI Thread and add some delays before connecting etc.

closed 2018-12-10 by roshanrajaratnam