nordicsemi/Android-nRF-Mesh-Library #108: .transport.LowerTransportLayer.sendBlockAck crash

WangManAndroid, 2018-12-13:
`12-13 10:43:30.429 7621-7621/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: discoverServices() - device: 98:12:88:77:00:15
12-13 10:43:30.429 7621-7643/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: onGetService() - Device=98:12:88:77:00:15 UUID=00001800-0000-1000-8000-00805f9b34fb
12-13 10:43:30.429 7621-7645/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: onGetService() - Device=98:12:88:77:00:15 UUID=00001801-0000-1000-8000-00805f9b34fb
12-13 10:43:30.429 7621-7696/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: onGetService() - Device=98:12:88:77:00:15 UUID=00001828-0000-1000-8000-00805f9b34fb
12-13 10:43:30.429 7621-7643/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: onGetCharacteristic() - Device=98:12:88:77:00:15 UUID=00002a00-0000-1000-8000-00805f9b34fb
12-13 10:43:30.439 7621-7645/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: onGetCharacteristic() - Device=98:12:88:77:00:15 UUID=00002a01-0000-1000-8000-00805f9b34fb
12-13 10:43:30.439 7621-7696/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: onGetCharacteristic() - Device=98:12:88:77:00:15 UUID=00002a04-0000-1000-8000-00805f9b34fb
12-13 10:43:30.439 7621-7643/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: onGetCharacteristic() - Device=98:12:88:77:00:15 UUID=00002a05-0000-1000-8000-00805f9b34fb
12-13 10:43:30.449 7621-7645/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: onGetCharacteristic() - Device=98:12:88:77:00:15 UUID=00002add-0000-1000-8000-00805f9b34fb
12-13 10:43:30.449 7621-7696/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: onGetCharacteristic() - Device=98:12:88:77:00:15 UUID=00002ade-0000-1000-8000-00805f9b34fb
12-13 10:43:30.449 7621-7643/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: onGetDescriptor() - Device=98:12:88:77:00:15 UUID=00002902-0000-1000-8000-00805f9b34fb
12-13 10:43:30.459 7621-7645/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: onGetDescriptor() - Device=98:12:88:77:00:15 UUID=00002902-0000-1000-8000-00805f9b34fb
12-13 10:43:30.459 7621-7696/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: onSearchComplete() = Device=98:12:88:77:00:15 Status=0
12-13 10:43:30.459 7621-7621/no.nordicsemi.android.nrfmeshprovisioner D/PRETTY_LOGGER: ┌────────────────────────────────────────────────────────────────────────────────────────────────────────────────
    │ Thread: main
    ├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄
    │ MeshProvisionerActivity$1.onChanged  (MeshProvisionerActivity.java:193)
    │    Debuger.d  (Debuger.java:16)
    ├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄
    │ Initializing…
    └────────────────────────────────────────────────────────────────────────────────────────────────────────────────
12-13 10:43:30.469 7621-7696/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: setCharacteristicNotification() - uuid: 00002ade-0000-1000-8000-00805f9b34fb enable: true
12-13 10:43:30.469 7621-7696/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: writeDescriptor() - uuid: 00002902-0000-1000-8000-00805f9b34fb
12-13 10:43:30.539 7621-7643/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: onDescriptorWrite() - Device=98:12:88:77:00:15 UUID=00002ade-0000-1000-8000-00805f9b34fb
12-13 10:43:30.539 7621-7643/no.nordicsemi.android.nrfmeshprovisioner D/PRETTY_LOGGER: ┌────────────────────────────────────────────────────────────────────────────────────────────────────────────────
    │ Thread: Binder_1
    ├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄
    │ MeshService.onDeviceReady  (MeshService.java:405)
    │    Debuger.d  (Debuger.java:16)
    ├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄
12-13 10:43:30.549 7621-7643/no.nordicsemi.android.nrfmeshprovisioner D/PRETTY_LOGGER: │ 98:12:88:77:00:15
    └────────────────────────────────────────────────────────────────────────────────────────────────────────────────
12-13 10:43:32.549 7621-7621/no.nordicsemi.android.nrfmeshprovisioner V/MeshTransport: Src address: 07FF
    Dst address: 002B
    Key: C3BE94232FFFB189256CC19760C5A4A0
    akf: 0
    aid: 0
    aszmic: 0
    Sequence number: 123
    Access message opcode: 32776
    Access message parameters: FF
12-13 10:43:32.549 7621-7621/no.nordicsemi.android.nrfmeshprovisioner V/AccessLayer: Created Access PDU 8008FF
12-13 10:43:32.549 7621-7621/no.nordicsemi.android.nrfmeshprovisioner V/UpperTransportLayer: Device nonce: 020000007B07FF002B00000000
12-13 10:43:32.559 7621-7621/no.nordicsemi.android.nrfmeshprovisioner V/UpperTransportLayer: Encrypted upper transport pdu: 7174AC29C3B4C1
12-13 10:43:32.559 7621-7621/no.nordicsemi.android.nrfmeshprovisioner V/LowerTransportLayer: Unsegmented Lower transport access PDU 007174AC29C3B4C1
12-13 10:43:32.559 7621-7621/no.nordicsemi.android.nrfmeshprovisioner V/NetworkLayer: Encryption key: 09D98362AFD9980568560D7ABB90472E
    Privacy key: F9CA93F9110E9AD5EF741B35DF2C35F4
    Sequence Number: 00007B
    Network nonce: 006400007B07FF000000000000
    Encrypted Network payload: 86FCB04510E4CDCBA38C2B8C6A44
12-13 10:43:32.559 7621-7621/no.nordicsemi.android.nrfmeshprovisioner V/ConfigCompositionDataGet: Sending composition data get
12-13 10:43:32.559 7621-7621/no.nordicsemi.android.nrfmeshprovisioner V/BleMeshManager: Sending data : 0x400C2793473B679886FCB04510E4CDCBA38C2B8C
12-13 10:43:32.559 7621-7621/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: writeCharacteristic() - uuid: 00002add-0000-1000-8000-00805f9b34fb
12-13 10:43:32.559 7621-7696/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: onCharacteristicWrite() - Device=98:12:88:77:00:15 UUID=00002add-0000-1000-8000-00805f9b34fb Status=0
12-13 10:43:32.569 7621-7696/no.nordicsemi.android.nrfmeshprovisioner V/BleMeshManager: Data written: 0x400C2793473B679886FCB04510E4CDCBA38C2B8C
12-13 10:43:32.569 7621-7621/no.nordicsemi.android.nrfmeshprovisioner V/BleMeshManager: Sending data : 0xC06A44
12-13 10:43:32.569 7621-7621/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: writeCharacteristic() - uuid: 00002add-0000-1000-8000-00805f9b34fb
12-13 10:43:32.569 7621-7643/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: onCharacteristicWrite() - Device=98:12:88:77:00:15 UUID=00002add-0000-1000-8000-00805f9b34fb Status=0
12-13 10:43:32.569 7621-7643/no.nordicsemi.android.nrfmeshprovisioner V/BleMeshManager: Data written: 0xC06A44
12-13 10:43:32.569 7621-7643/no.nordicsemi.android.nrfmeshprovisioner V/MeshManagerApi: Network pdu sent: 0x000C2793473B679886FCB04510E4CDCBA38C2B8C6A44
12-13 10:43:32.569 7621-7643/no.nordicsemi.android.nrfmeshprovisioner V/MeshMessageHandler: Switching current state on write complete ConfigCompositionDataGet to ConfigCompositionDataStatus
12-13 10:43:32.639 7621-7645/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: onNotify() - Device=98:12:88:77:00:15 UUID=00002ade-0000-1000-8000-00805f9b34fb
12-13 10:43:32.639 7621-7645/no.nordicsemi.android.nrfmeshprovisioner V/BleMeshManager: Characteristic notified: 0x400C3BF3E4D5D32B6611C9A0430168E45E013C5C
12-13 10:43:32.679 7621-7696/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: onNotify() - Device=98:12:88:77:00:15 UUID=00002ade-0000-1000-8000-00805f9b34fb
12-13 10:43:32.679 7621-7696/no.nordicsemi.android.nrfmeshprovisioner V/BleMeshManager: Characteristic notified: 0xC08528CB43E9BF1DECC4B0
12-13 10:43:32.679 7621-7696/no.nordicsemi.android.nrfmeshprovisioner V/MeshManagerApi: Received network pdu: 0x000C3BF3E4D5D32B6611C9A0430168E45E013C5C8528CB43E9BF1DECC4B0
12-13 10:43:32.689 7621-7696/no.nordicsemi.android.nrfmeshprovisioner V/NetworkLayer: Privacy Random: 0000000000000000006611C9A0430168
    TTL for received message: 10
    Sequence number of received access message: 0
    Received a segmented access message from: 002B
[... 70 lines cut ...]
12-13 10:43:33.399 7621-7696/no.nordicsemi.android.nrfmeshprovisioner V/ConfigCompositionDataStatus: Number of sig models: 2
    Number of vendor models: 0
    Sig model ID 0 : 0000
    Sig model ID 1 : 1000
    Location identifier: 0100
    Number of sig models: 1
    Number of vendor models: 0
    Sig model ID 0 : 1000
    Number of elements: 2
[... cut ...]
        at android.os.Looper.loop(Looper.java:136)
        at android.app.ActivityThread.main(ActivityThread.java:5052)
        at java.lang.reflect.Method.invokeNative(Native Method)
        at java.lang.reflect.Method.invoke(Method.java:515)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:796)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:612)
12-13 10:43:34.059 7621-7621/no.nordicsemi.android.nrfmeshprovisioner W/System.err:     at dalvik.system.NativeStart.main(Native Method)`

WangManAndroid, 2018-12-13:
It may appear on some Android phones

roshanrajaratnam, 2018-12-13:
@WangManAndroid Is it possible to provide more details on this? This is something I haven't seen and have you made any changes?

Also the library have moved on from using the version 1.0.4 library and the latest dev branch contains a lot of the new improvements and bug fixes. I highly recommend to switch to the newer version as it has fixed a lot of the internals in the library.

roshanrajaratnam, 2019-02-06:
Closing this due to inactivity. Please feel free to reopen!

closed 2019-02-06 by roshanrajaratnam