nordicsemi/Android-nRF-Mesh-Library #109: sometimes TID remains same if controlling Group of NODEs

vikrant8051, 2018-12-13:
As per Mesh Model Specification (on firmware end) ,
```

if ( TID ==  LAST_TID &&
     SRC_ADDR == LAST_SRC_ADDR &&
     DST_ADDR == LAST_DST_ADDR &&
     (now - last_msg_timestamp <= 6 Seconds) ) {

  // re-transmitted message & hence
  return;

}
```
**Android #nRFMesh issue:**
If App is controlling Group of NODEs which actually has only one NODE then each button toggle event send message with new TID.  This is perfect implementation.

But it is observed that if there are many NODEs actually available then APP sends message with same TID. This happens intermittently. Because of it, above mentioned condition get executed on firmware end & server assumes it is re-transmitted message & state does not get updated.

vikrant8051, 2018-12-13:
Mentioned issue does not arise with any other Vendor App as well as with iOS version of nRFMesh.

roshanrajaratnam, 2018-12-13:
Hi @vikrant8051 Thanks for reporting this and yes there was a bug in the TID when it comes to unacknowleged messages. However this was fixed internally in the current dev version.

vikrant8051, 2018-12-13:
@roshanrajaratnam Thanks for acknowledging the issue.
 
unacknowledged message ?
FYI, App (which is available on Google play store) always sends acknowledged message no matter what is Destination address.

Ideally, if Destination address is Group address then client should send unack message.

roshanrajaratnam, 2018-12-18:
@vikrant8051 sorry my bad. Currently the app uses the sequence number as the TID. In case an acknowledged message is lost you might notice the same sequence number in the next messages. Also there were some other improvements made internally that have fixed this.

roshanrajaratnam, 2018-12-18:
Closing in favor of #113

closed 2018-12-18 by roshanrajaratnam