nordicsemi/Android-nRF-Mesh-Library #111: MeshMessages like GenericOnOffSet should not have to pass a node.

DomiR, 2018-12-13:
Why do we want to pass the node when generating new messages like `GenericOnOffSet(NODE,...,destinationAddress)`. When sending to groups we do not have a single node we can put there. Can we just remove this from all model messages, or why is it necessary?

roshanrajaratnam, 2018-12-14:
@DomiR you have a very good point. The only reason is or rather was, almost everything was stored in the node object so I had to pass around the node object. However now after separating the mesh network properties and node properties don't think we need to pass it around anymore. Currently I'm finishing up config relay set and I will look in to this and clean that up as well.

roshanrajaratnam, 2018-12-18:
@DomiR I have started looking in to this issue, I also think none of the messages require the node to be passed. I will make the much needed changes, although this means the `sendMeshConfigurationMessage `and `sendMeshApplicationMessage` in the MeshManagerApi will need to have the src and dst addresses in the signature.

DomiR, 2018-12-18:
True, we can let us inspire by https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/mesh-api.txt`.
They have:
`void Send(object element_path, uint16 destination, uint16 key_index, array{byte} data)` 
where `element_path` is the local client model, which we are currently not using, we could use the sourceAddress here. `key_index` is looking up the application key index of the local client model, we could look it up in the network settings (global app key index). `data` is the whole Payload, we could name this as api `sendMeshRawMessage` to give full control to developer (build messages themself) and have `sendMeshApplicationMessage` assemble the payload and use `sendMeshRawMessage` internally.

roshanrajaratnam, 2018-12-18:
I will take a look and see what I can use out of this.

DomiR, 2018-12-18:
Alright, let me know, where I can help 😄

roshanrajaratnam, 2018-12-18:
Thanks you sir! I'm on it already, some refactoring here :/

roshanrajaratnam, 2018-12-18:
@DomiR I think I have managed to refactor the code to remove the node from all messages. I have pushed the changes to the following branch improvements/remove-node-from-generic-messages-#111.
I have done a quick testing may be you can give it a test to see if everything works as expected?

roshanrajaratnam, 2018-12-19:
@DomiR Basically made a MINOR mess out of it ;) I believe this should solve your issue :)

PR #127 (2019-01-16): Preparing for release v1.1.0
* Brand new Nordic Identity colors!!!!
* Fixes #73
  - Added message classes and removed states for status messages.
* Fixes #32 
  - Added server configuration messages such as NetworkTransmit, ProxyState and Relay.
* Fixes #36 
  - Supports importing and exporting mesh networks. This allows users to share network across nRF Mesh Apps.
  - Added RoomDB to mesh library replacing shared preference storage. 
* Fixes #40 
  - Library supports deleting nodes.
* Fixes #61 
  - Selects a node based on the node address which includes element address. In case of a certain  [... cut]
closed 2019-01-16 by roshanrajaratnam