nordicsemi/Android-nRF-Mesh-Library #115: Wrong nomenclature for paramenters under Retransmission

vikrant8051, 2018-12-18:
@roshanrajaratnam @mostafaberg 

Publish(under every Model) -> SET PUBLICATION -> Retransmission 
...... has issue.

Here under Retransmission there are two sub-parameters:

    #RetransmitCount (#RC)
    #IntervalSteps (#IS) 

When user set #RC & #IS values via App, at firmware level those get assign to #IS & #RC.

Means nomenclature is wrong at APP level.

roshanrajaratnam, 2018-12-18:
@vikrant8051 Thanks for reporting this. Will look and get back

roshanrajaratnam, 2018-12-18:
Issue confirmed, sorry about this, the parameters were swapped in the message. Will push the changes later!

vikrant8051, 2018-12-19:
Please inform about it to iOS team because same issue is seen under iOS App.

roshanrajaratnam, 2018-12-19:
Please create an issue on the relevant repository for this. This makes it easier to keep track of things :) Thanks

PR #116 (2018-12-19): Bugfix/config relay set params #115
* Fixes #115 
  - Fixes a bug causing the ConfigRelaySet to send retransmission count and interval steps to be swapped
  - Fixes a bug in the ConfigurationServerModel UI displaying interval steps
* Fixes an additional issue causing the node features to be parsed incorrectly
PR #127 (2019-01-16): Preparing for release v1.1.0
* Brand new Nordic Identity colors!!!!
* Fixes #73
  - Added message classes and removed states for status messages.
* Fixes #32 
  - Added server configuration messages such as NetworkTransmit, ProxyState and Relay.
* Fixes #36 
  - Supports importing and exporting mesh networks. This allows users to share network across nRF Mesh Apps.
  - Added RoomDB to mesh library replacing shared preference storage. 
* Fixes #40 
  - Library supports deleting nodes.
* Fixes #61 
  - Selects a node based on the node address which includes element address. In case of a certain  [... cut]
closed 2019-01-16 by roshanrajaratnam