nordicsemi/Android-nRF-Mesh-Library #122: Binding key and publishing/subscribing health model

DuyTrandeLion, 2018-12-28:
I used the nRF Mesh app to config the health server to publish to a specific health client, they are all binded with the same app key. I use the app as the provisioner, but the health client doesn't receive any message. 

I don't know if the nRF Mesh app or library support binding for health server/client or it's a bug, can it be supported or fixed? Can I fixed it myself? I am not very good at Android developing.

Thank you,
Best regards,
    Duy

roshanrajaratnam, 2019-01-02:
Hi @DuyTrandeLion sorry for the late reply as I was on vacation. So far the app only supports key binding message on the health server. Did you have trouble binding app keys?

DuyTrandeLion, 2019-01-04:
Hi @roshanrajaratnam ,
Thank you for your feedback, I don't have problem binding the key, but I don't know where to add the code for period configuration. I wonder when will Nordic support it?

roshanrajaratnam, 2019-01-04:
@DuyTrandeLion what do you mean by period configuration?

DuyTrandeLion, 2019-01-04:
@roshanrajaratnam my bad, I mean the publishing period.

roshanrajaratnam, 2019-01-16:
Closing this issue since there is no updates on this!

closed 2019-01-16 by roshanrajaratnam