nordicsemi/Android-nRF-Mesh-Library #128: Invalid TransactionID being used when sending messages to group addresses

roshanrajaratnam, 2019-01-17:
Incorrect transactioin ids are used when sending messages to group addresses. This is due to the fact that the last received sequence number from a node is being used as the transaction Id when sending application layer messages. However when it comes to group addresses a single node's sequence number cannot be used as a transaction id.

DomiR, 2019-01-25:
What is the plan here? Should we just have a global counter for TIDs? They are valid for only several seconds, so it should not be a problem to use a global counter that overflows... We are using one in our app... The api to set custom TID should stay there though, as we are currently running into problems in high noise environments (>58 wifi nets surrounding us). We need to resend messages with the same TID...

roshanrajaratnam, 2019-01-25:
Well this is more of an issue related to the app side and using a sequence number of  a particular node as a TID to a group address. However its fixed for now by using the sequence number of the provisioner which increments at all times and also can be used for re-sending as well. I won't remove the custom api. I'll leave it for users to set their own like you had suggested before and it's more convenient that way I guess.

PR #155 (2019-02-12): Improvements/add group controlling #129
* Fixes #129,  #41 
  - Replaces the scanner view with a group view. In order to scan/provision a device, use the add floating action button.
  - Group view adds support for creating/editing a group. When subscribing a model to a group a user may create or select an existing group
  - Model Configuration view now only sends messages to unicast addresses.
* Fixes #128 
  - The app uses the provisioner sequence number as the tid
* Fixes #153
  -  Fixed invalid endianness 
* Fixes #140 
  - Fixed updating a hardcoded key
* Fixes #139 
  - Added missing `not [... cut]
closed 2019-02-12 by roshanrajaratnam