nordicsemi/Android-nRF-Mesh-Library #132: ProvisionedMeshNode Parcelable implementation broken

HaukeRa, 2019-01-22:
---

**Describe the bug**
Generally Parcelable objects are expected to contain the same information before and after parceling them and retreiving them from a parcel.

Two issues:
1. ProvisionedMeshNode(Parcel in):
`110:  isProvisioned = in.readByte() != 1;
 111: isConfigured = in.readByte() != 1;
 137: blackListed = in.readByte() != 1`
inverts the state written to the parcel (written as true --> 1, false -->0. read as 1-->false, 0-->true)
2. Deprecated fields not included in parcelable (not so bad, as they are deprecated anyways)
Example: NetworkKey in BaseProvisionedMeshNode is always null after reading ProvisionedMeshNode from parcel

**Expected behavior**
Contents of objects equal before and after parceling/unparceling

HaukeRa, 2019-01-23:
Another error:

* blackListed is written with Parcel.writeValue() but read with Parcel.readByte()

This results in an error in SOME cases. I spent all day trying to write a test for this, just now I found a way to test this.

While it works in most cases, sometimes it results in a hard to debug crash. Internally Parcel.writeValue writes two Integers, one as a flag for what type has been written and one for the value itself. Parcel.writeByte only writes the value itself. So using not the corresponding method results in a partially read Parcel, so subsequent reads might fail with weird errors like: `java.lang.RuntimeException: Parcel android.os.Parcel@3ed21c6: Unmarshalling unknown type code 7209057 at offset 2012.`

roshanrajaratnam, 2019-01-23:
Nice catch, ideally it shouldn't be writeValue at all.

roshanrajaratnam, 2019-01-24:
True, but this is something wrong, it shouldn't be a Nullable value at all. Fixed internally! Thanks for pointing out!

PR #155 (2019-02-12): Improvements/add group controlling #129
* Fixes #129,  #41 
  - Replaces the scanner view with a group view. In order to scan/provision a device, use the add floating action button.
  - Group view adds support for creating/editing a group. When subscribing a model to a group a user may create or select an existing group
  - Model Configuration view now only sends messages to unicast addresses.
* Fixes #128 
  - The app uses the provisioner sequence number as the tid
* Fixes #153
  -  Fixed invalid endianness 
* Fixes #140 
  - Fixed updating a hardcoded key
* Fixes #139 
  - Added missing `not [... cut]
closed 2019-01-24 by roshanrajaratnam