nordicsemi/Android-nRF-Mesh-Library #134: More broken parcelable implementations

HaukeRa, 2019-01-23:
---

**Describe the bug**
Some meshprovisioner.transport.MeshMessage subclasses implement Parcelable.
They implement writeToParcel as: `dest.writeValue(mMessage)` whre mMessage is a transport.Message.
But Message does not implement Parcelable so attemptint to parcelize a MeshMessage always fails.

Affected classes:
- GenericOnOffStatus
- GenericLevelStatus
- ConfigCompositionDataStatus
- ConfigModelPublicationStatus
- ConfigNodeResetStatus
- ConfigModelSubscriptionStatus
- VendorModelMessageStatus
- ConfigAppKeyStatus
- ConfigModelAppStatus
- SceneStatus
- ConfigRelayStatus
- LightLightnessStatus
- ConfigNetworkTransmitStatus
- LightCtlStatus
- LightHslStatus
- SceneRegisterStatus
- ConfigProxyStatus 

**To Reproduce**
Steps to reproduce the behavior:
1. Try to parcelize any of the listed classes

**Logs / Screenshots**
Example Stack trace: 
```
java.lang.RuntimeException: Parcel: unable to marshal value no.nordicsemi.android.meshprovisioner.transport.AccessMessage@5781ffb8
	at android.os.Parcel.writeValue(Parcel.java:1656)
	at no.nordicsemi.android.meshprovisioner.transport.LightHslStatus.writeToParcel(LightHslStatus.java:148)
	at android.os.Parcel.writeParcelable(Parcel.java:1675)
```

HaukeRa, 2019-01-23:
**MeshModel.java**:
`try{]catch()` block in Constructor from Parcel can result in the same errors as I mentioned in the ProvisionedMeshNode issue. On an exception the parcel remains partially read, so any data after your own is offset now, resulting in wrong data or hard to debug crashes.

roshanrajaratnam, 2019-01-24:
That was a redundant `try{]catch()`. If i recall correctly this was added to debug a app crash but not being removed after. Fixed internally.

HaukeRa, 2019-01-24:
Did you account for all the broken MeshMessages in the original bug report?

PR #161 (2019-02-20): Bugfix/broken parcelable implementation #134
* Fixes #134 
  - Added missing parcelable implemenation in `AccessMessage` and `ControlMessage` classes
  - Fixed broken parcelable implementation  on status message classes
* Adds missing company identifiers
PR #172 (2019-03-05): Release v1.2.0
* Fixes #143,  
* Fixes #129, #41 
  - Adds support for controlling groups. Previously this was done in each model configuration view but now model configuration will only allow sending messages to the unicast address itself.
* Fixes #159 
  - Fixes a documentation bug after renaming callbacks.
* Fixes #149 
  - All addresses now use ints instead of byte arrays.
* Fixes #134 
  - Fixes a bug relating to broken parcelable implementations
* Fixes #169 
  - Added an additional api to override default 5 second attention timer.
`identifyNode(@NonNull final UUID deviceUuid, final Stri [... cut]
closed 2019-03-05 by roshanrajaratnam