nordicsemi/Android-nRF-Mesh-Library #135: Provisoning after network reset stucks at "Sending composition data get"

PopradiArpad, 2019-01-23:
---

**Describe the bug**
Provisoning after network reset stucks at "Sending composition data get",
after app restart provisioning works again. 

**To Reproduce**
0. Build BLE Mesh Light Switch Server Demo from nRF5 SDK for Mesh 3.1.0 with SDK 15.2
    at Segger Embedded Studio, connect a DK to the PC, erase all, run.
1. Start nRF Mesh 1.1.0, no network is known
2.Go to 'Scanner'
3. Click on the device.
4. Click on 'IDENTIFY'.
5. Click on 'PROVISION'.
6. Configuration Complete. Click on 'OK'
7. Click on 'DISCONNECT'. //optional
8. Click on 'Settings'.
9. Click on 'RESET'.
10. Click on 'YES'.
11. Go to Segger, stop sw run, connect, erase all, run
12 Go to nRF Mesh 1.1.0
13.Go to 'Scanner'
14. Click on the device.
15. Click on 'IDENTIFY'.
16. Click on 'PROVISION'.
17. The Provisioning stucks at "Sending composition data get..."

Log on the device site with higher log level:
`
static void initialize(void)
{
//    __LOG_INIT(LOG_SRC_APP | LOG_SRC_ACCESS | LOG_SRC_BEARER, LOG_LEVEL_INFO, LOG_CALLBACK_DEFAULT);
    __LOG_INIT(LOG_SRC_APP | LOG_SRC_ACCESS | LOG_GROUP_STACK | LOG_SRC_PROV, LOG_LEVEL_INFO, LOG_CALLBACK_DEFAULT);
    __LOG(LOG_SRC_APP, LOG_LEVEL_INFO, "----- BLE Mesh Light Switch Server Demo -----\n");
.....
}

<t:          0>, main.c,  239, ----- BLE Mesh Light Switch Server Demo -----
<t:      16162>, main.c,  218, Initializing and adding models
<t:      16165>, main.c,  113, App OnOff Model Handle: 2
<t:      21011>, prov_bearer_adv.c,  400, PB-ADV: context at 0x20002F68 added to bearer
<t:      21022>, mesh_app_utils.c,   65, Device UUID (raw): 2957DB05A74C4E4EAF8439B3CF773732
<t:      21026>, mesh_app_utils.c,   70, Device UUID : 05DB5729-4CA7-4E4E-AF84-39B3CF773732
<t:    1942292>, ble_softdevice_support.c,  104, Successfully updated connection parameters
<t:    2007865>, ble_softdevice_support.c,  104, Successfully updated connection parameters
<t:    2899385>, main.c,  198, Successfully provisioned
<t:    2899389>, main.c,  209, Node Address: 0x0001 
<t:    2980236>, proxy.c,  579, Connected
<t:    3002538>, ble_softdevice_support.c,  104, Successfully updated connection parameters
<t:    3067331>, ble_softdevice_support.c,  104, Successfully updated connection parameters
<t:    3097881>, proxy.c,  606, TX ready
<t:    3130317>, proxy.c,  619, TX complete
<t:    3187095>, proxy.c,  611, RX
<t:    3187101>, proxy.c,  553, RX GATT PDU type 0x0, len 21
<t:    3187107>, net_packet.c,  230, Unencrypted data: : 000100C65065E36E3411
<t:    3187113>, transport.c,  934, Message decrypted
<t:    3187138>, transport.c, 1233, TX:SAR packet: 95B8CABA4BEC33B73E4B7CAF7CFDCAAE366B934B8AF2B77081A3
<t:    3203307>, proxy.c,  619, TX complete
<t:    3219527>, proxy.c,  619, TX complete
<t:    3235747>, proxy.c,  619, TX complete
<t:    3251967>, proxy.c,  619, TX complete
<t:    3268189>, proxy.c,  619, TX complete
<t:    3284407>, proxy.c,  619, TX complete
<t:    3300627>, proxy.c,  619, TX complete
<t:    3301889>, transport.c,  495, Dropped SAR session 0, reason 1
<t:    3316847>, proxy.c,  619, TX complete
<t:    3333068>, proxy.c,  619, TX complete
<t:    3349288>, proxy.c,  619, TX complete
`

This behavior is reproducible and the same with SDK for Mesh 2.2.0, SDK 15.0, nRF Mesh 1.1.0 

**Expected behavior**
The provisioning after network reset should work like the one before network reset without restarting the app.

**Platform details:**
 - Device: Samsung S7, Huawei idontknow
 - OS: Android 8.0.0 at S7
 - nRF Mesh 1.1.0

The DK reset shows that the problem not on the DK side. It happens even with DK sw build with SDK for Mesh 2.2.0. With some extra log to see, that the handle_composition_data_get is called and it tries to send replies unsuccessfully (reason 1 is NRF_MESH_SAR_CANCEL_REASON_RETRY_OVER).

PopradiArpad, 2019-01-23:
Without **network reset** the second device can be provisioned.

roshanrajaratnam, 2019-01-23:
Hi @PopradiArpad, thanks for reporting this issue. Is there any chance you could try this with the precompiled examples?

PopradiArpad, 2019-01-24:
Hi @roshanrajaratnam, thanks for the quick reply. Where can I find the precompiled examples? They are not supplied with the code anymore.

roshanrajaratnam, 2019-01-24:
@PopradiArpad If you download the mesh sdk 3.1.0 and unzip the file, you can find all the precompiled hex files in the following path nrf5SDKforMeshv310src\bin\merged\ospace\examples.

PopradiArpad, 2019-01-24:
@roshanrajaratnam . The same problem.
I have used
`nrfjprog -f nrf52 -e`
`nrfjprog -f nrf52 --program light_switch_server_nrf52832_xxAA_s132_6.1.0_merged_sd.hex --verify --sectorerase`

roshanrajaratnam, 2019-01-25:
@PopradiArpad sorry for the delay, Support issues like these should ideally go to the DevZone. Please be kind to open a ticket on our DevZone page. In the mean time I'll try to find some time to have a look at this. The chances of  getting help via DevZone is much higher though! Thanks for understanding!

roshanrajaratnam, 2019-02-06:
Closing this in favor of Devzone ticket.

closed 2019-02-06 by roshanrajaratnam