nordicsemi/Android-nRF-Mesh-Library #137: Database inconsistencies

HaukeRa, 2019-01-23:
---

**Describe the bug**
Some properties are annotated with Ignored so that they are not persisted in the database. Some of them however are actually set during creation/provisioning etc.

One example is: isProvisioned in BaseProvisionedMeshNode. It is copied from UnprovisionedMeshNode (where it is set to true during the provisioning process).

When persisted and then loaded from database, these fields take their default values, in this case it results in the unexpected behaviour that a `ProvisionedMeshNode` has a propertiy `isProvisioned == false`

**To Reproduce**
Steps to reproduce the behavior:
1. Provision node
2. see that `isProvisioned` is true for `ProvisionedMeshNode`
3. Restart the App
4. See that `isProvisioned` is false for `ProvisionedMeshNode`

**Expected behavior**
Objects should be fully persisted, unused Fields should be deprecated or better removed.
When a field is not persisted, it should be a computed property (computed from other properties)

**Platform details:**
 - Library Version: 1.1.0 Release

roshanrajaratnam, 2019-01-24:
Hi @HaukeRa  thanks for reporting this. This has somehow escaped the refactoring that was done for release 1.1.0.

roshanrajaratnam, 2019-01-24:
Fixed internally

PR #155 (2019-02-12): Improvements/add group controlling #129
* Fixes #129,  #41 
  - Replaces the scanner view with a group view. In order to scan/provision a device, use the add floating action button.
  - Group view adds support for creating/editing a group. When subscribing a model to a group a user may create or select an existing group
  - Model Configuration view now only sends messages to unicast addresses.
* Fixes #128 
  - The app uses the provisioner sequence number as the tid
* Fixes #153
  -  Fixed invalid endianness 
* Fixes #140 
  - Fixed updating a hardcoded key
* Fixes #139 
  - Added missing `not [... cut]
closed 2019-01-24 by roshanrajaratnam