nordicsemi/Android-nRF-Mesh-Library #138: Replacing App Key inserts new key instead of replacing

HaukeRa, 2019-01-24:
---

**Describe the bug**
When trying to replace an app key at a specific index (`BaseMeshNode.public void addAppKey(final int keyIndex, final ApplicationKey newAppKey)`), the method comments read:
_Adds an app key to the list of keys with the given key index. If there is an existing key with the same index, it will be replaced._

Internally the call is delegated to a Java `List.add` with index. It behaves as:
`Inserts the specified element at the specified position in this list (optional operation). Shifts the element currently at that position (if any) and any subsequent elements to the right (adds one to their indices).`

So the new key is actually inserted into the list and all other keys are shifted to the right

Additionally a List for the keys is probably not the best Container? When the keys are shifted, the indices in the list do not match up with the ApplicationKey.keyIndex property. I don't know about the consequences of this, but it doesn't seem right.

roshanrajaratnam, 2019-01-25:
@HaukeRa thanks! the comment is what's wrong here, hence misleading. Same app key should not be allowed to be added. Also there is a separate method to update an app key I'll update the documentation.

roshanrajaratnam, 2019-01-25:
Fixed internally!

closed 2019-01-25 by roshanrajaratnam

HaukeRa, 2019-01-25:
Did you consider not using `List` as a container for the Keys? It is very error prone or at least misleading (in a sense that it is not clear whether the indices in the list really matter), that the index of the list and the key are connected in a way.

This means you have double the work ensuring that both match.

In general, how would it be handled if you have three keys, delete the first one, should the indices of the others be adapted? Should they stay the same and key index 0 is free for further use?

Also are you planing to push the fixes to the dev branch (or some other branch) so I can rebase on them?

roshanrajaratnam, 2019-01-25:
Hi I see what you mean. The initial thinking was to have a list with ApplicationKeys where the index within the ApplicationKey is what dictates so that the indexes in the list do not matter. The indices in the list does not matter this way. If you Look in the example app Settings >> Manage App Keys >> deleting the first key in the list does not change the index. So index 0 is still available for use. Currently i've done the fixes in a local branch but will push them early next week.  I've done the smaller fixes in the local branch but probably will have to spend some time fixing the other parcelable issues you mentioned separate.

HaukeRa, 2019-01-25:
Okay I understand. But then why bother placing the Key at the supplied keyIndex? I would go for removing the explicit separate keyIndex from the API alltogether.

And then managing the List (or some other container) within the BasheMeshNetwork class. Sorting it by the `keyIndices` of the `ApplicationKeys` seems appropriate.

roshanrajaratnam, 2019-01-25:
The idea was that a key could be placed in any index of the list, but this is actually not validated at all, given that someone tries to place it in an index larger than the size. However looking back not sure if its useful either...

HaukeRa, 2019-01-25:
When the list index has no iternal use (meaning it is not related to the keyIndex in any way that the stack uses this ordering) it is solely a presentation issue how the keys are ordered. In that way I would order the keys internally after their inherent index. If the UI needs a different ordering for presentation, it easily can sort them as needed in the UI (not affecting the stack).

roshanrajaratnam, 2019-01-28:
Yeah makes sense, when I was implementing the app it was easier to have it inserted at the specified index, but like you said probably makes sense to sort it within the app and maintain the insertion order.

PR #155 (2019-02-12): Improvements/add group controlling #129
* Fixes #129,  #41 
  - Replaces the scanner view with a group view. In order to scan/provision a device, use the add floating action button.
  - Group view adds support for creating/editing a group. When subscribing a model to a group a user may create or select an existing group
  - Model Configuration view now only sends messages to unicast addresses.
* Fixes #128 
  - The app uses the provisioner sequence number as the tid
* Fixes #153
  -  Fixed invalid endianness 
* Fixes #140 
  - Fixed updating a hardcoded key
* Fixes #139 
  - Added missing `not [... cut]