nordicsemi/Android-nRF-Mesh-Library #139: First App Key not persisted

HaukeRa, 2019-01-24:
---

**Describe the bug**
When Adding App Keys, the branch that is taken when the list of ApplicationKeys is empty does not call `notifyAppKeyAdded(newAppKey)`, resulting in it not beeing persisted in the Room Database. It transiently stays around though, so it only disappears when the app is completely restarted

**To Reproduce**
Steps to reproduce the behavior:
1. Start with an MeshNetwork without any app keys (or delete all)
2. Add one or more App Keys
3. Restart the App (make sure to remove it from the task list)
4. See first app key is missing

**Expected behavior**
All ApplicationKeys being persisted

roshanrajaratnam, 2019-01-25:
@HaukeRa Thanks for this. Your testing really helps!

roshanrajaratnam, 2019-02-06:
Fixed internally.

PR #155 (2019-02-12): Improvements/add group controlling #129
* Fixes #129,  #41 
  - Replaces the scanner view with a group view. In order to scan/provision a device, use the add floating action button.
  - Group view adds support for creating/editing a group. When subscribing a model to a group a user may create or select an existing group
  - Model Configuration view now only sends messages to unicast addresses.
* Fixes #128 
  - The app uses the provisioner sequence number as the tid
* Fixes #153
  -  Fixed invalid endianness 
* Fixes #140 
  - Fixed updating a hardcoded key
* Fixes #139 
  - Added missing `not [... cut]
closed 2019-02-06 by roshanrajaratnam