nordicsemi/Android-nRF-Mesh-Library #141: BaseMeshNetwork.removeAppKey not working under all circumstances

HaukeRa, 2019-01-24:
---

**Describe the bug**
BaseMeshNetwork.removeAppKey(ApplicationKey) internally calls the Java List API .remove(..).
This API relies on an .equals() implementation to remove an item. ApplicationKey does not implement .equals() so this falls back to Object.equals() which just checks reference equality.

As ApplicationKey is also a Parcelable, multiple copies of the same key (meaning same keyIndex and byte[]) may exist.

More specifically it is now impossible to remove an ApplicationKey if it has been parcelled and then unparcelled.

roshanrajaratnam, 2019-02-06:
@HaukeRa Unfortunately i'm not able to reproduce this issue. Deleting an app key from the manage app keys option in the settings lets me remove an app key.

roshanrajaratnam, 2019-03-05:
Closing due to inactivity here, please feel free to open this again.

closed 2019-03-05 by roshanrajaratnam

HaukeRa, 2019-08-24:
Sorry for the inactivity. ApplicationKey/MeshKey by now implements equals() in a compatible way.
This should resolve this issue.