nordicsemi/Android-nRF-Mesh-Library #144: provide library through gradle implementation 'no.nordicsemi.android:...

gridar, 2019-01-30:
Hi Nordic, thanks for this repo it's great.
Is there a way to easily add the mesh library as dependency ? like `implementation 'no.nordicsemi.android:ble:1.2.0'`
for now the doc say we need to add :
`project(':meshprovisioner').projectDir = file('../../Android-nRF-Mesh-Library/android-nrf-mesh-library/meshprovisioner')` in the settings.gradle. That's working fine but it would be great to have it without cloning the repo on locale machines.

roshanrajaratnam, 2019-01-31:
Hi @gridar, currently we haven't published to jcenter yet. Something we would do sometime in the future. Hope this helps

closed 2019-02-05 by roshanrajaratnam

gridar, 2020-01-06:
Hello @roshanrajaratnam, any news on putting the lib in jcenter ? That'll help us making a flutter plugin out of it :)

roshanrajaratnam, 2020-01-06:
@gridar sorry about the delay, been a bit busy with some other priorities. I will take a look at this tomorrow and try to get a version on jcenter.

gridar, 2020-01-27:
Hello @roshanrajaratnam , just a friendly reminder about jcenter 😃

Kleak, 2020-05-18:
i think this issue should stay open until you publish the sdk on a package manager.

philips77, 2020-05-18:
It's there already! Since last week! Sorry, we forgot to update this issue.

Kleak, 2020-05-19:
awesome thanks

philips77, 2020-05-19:
Mind that package names have changed, simplified.