nordicsemi/Android-nRF-Mesh-Library #145: JSON files can only be loaded from nrf Mesh directory

POHakansson, 2019-01-31:
---

**Describe the bug**
When trying to load a JSON network config file, it only seems to work if the file's location is the Nordic Semiconductor / nrf Mesh directory.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the Settings tab and chose Import from the menu
2. Select a file from for example the "Downloads" folder of your phone
3. The error message says "Invalid file type"

**Expected behavior**
It should be possible to import a network from any directory location.

**Platform details:**
 - Device: Galaxy S7 and Galaxy S8
 - OS: Android 8.0.0
 - 1.1.0 January 16th from Play Store

roshanrajaratnam, 2019-02-06:
@POHakansson This does not seem to be the case when testing against the same device I have. The json file seems to get imported irrespective of the location. When you say select a file, are you trying to select any json file? It must be a json file or format exported from the same app (different phone).

roshanrajaratnam, 2019-02-08:
@POHakansson any updates on this?

POHakansson, 2019-02-08:
I must have made the same mistake several times when I decided to create an issue. Now when I'm testing, it works OK.

Sorry bout that,
Pär-Olof

closed 2019-02-08 by POHakansson

roshanrajaratnam, 2019-02-08:
No problem :)

i-am-smirnoff, 2019-02-14:
Hello!
Let me concretize the issue which exists.

Everything works fine if you navigate to file for import via device internal storage file structure. But if you select a file in "Downloads" as mentioned or if you select a file from "Drive" storage then 'Invalid file type' error occures.

I've managed to import 'invalid' file from "Drive" after changing condition in `if` to `true` here: 
_Android-nRF-Mesh-Library/android-nrf-mesh-library/meshprovisioner/src/main/java/no/nordicsemi/android/meshprovisioner/MeshManagerApi.java_

Currently it's line 807:

    if (uri.getPath().endsWith(".json")) {

**Platform details:**

* Device: Lenovo TB-8304F1
* OS: Android 7.0
* Languages: English (United States)
* 1.1.0 January 16th from Play Store