nordicsemi/Android-nRF-Mesh-Library #146: Impossible to re-use unicast address

POHakansson, 2019-01-31:
During provisioning of one of my 30+ office nodes the sequence was interrupted by some type of error on the node to be provisioned. I reset the node to be provisioned and intended to redo the provisioning using the address that I had allocated for that specific node. This used to be possible in older versions of the app, like the really old 0.2 from 2018 I've long been using.

I understand that preventing the user from re-using a "known" unicast address is intended to be a helping hand, but for me it proved a big problem since I needed the node to be have that specific address and there seems to be no method of removing a node from the network except connecting to it and doing "Reset node" from the app. Which is not possible if the node is not properly provisioned and can be connected to...

**To Reproduce**
Steps to reproduce the behavior:
1. Provision a node with any specific unicast address, e.g. 0x0126
2. Try provisioning another node with address 0x0126
3. The app ignores attempts to set next unicast address to 0x0126

**Expected behavior**
The app accepts that the user specifically sets a new unicast address.

**Platform details:**
 - Device: Galaxy S7, Galaxy S8
 - OS: Android 8.0.0
 - 1.1.0 from Play Store

roshanrajaratnam, 2019-02-06:
Hi @POHakansson, in the new version we introduced the device uuid as the unique which is the correct way to go instead of using the device uuid. Like you mentioned, the way to re-use an address is by resetting an existing node and re-using that address. You should also be careful when doing this so that the address do not overlap with another device. 

Coming back to your issue, what do you mean by if the device is not properly provisioned? If its not properly provisioned i assume its still in an unprovisioned state allowing you to provision and the app does connect to any provisioned node. 

Ideally according to mesh specification section (3.10.7) re-using an address is only allowed after removing(reset) that node from a network and then doing a key refresh procedure, but here as in this case we allow the user to do it after resetting a node that is using this address.  So the only way to do this as of now is only by resetting the node using the address.

closed 2019-02-06 by roshanrajaratnam

POHakansson, 2019-02-06:
Hi

Thanks for your reply. I guess I didn't consider the security aspects of re-using an address.

In my network my 35 nodes (and growing)  have a specific number and location in the office. We have a node-RED application which logs values and when I'm forever updating the nodes I want to re-use the old addresses.

During my development there are many cases when provisioning fails or is partial (my code, the Nordic mesh SDK v2.2.0 asserting half way through) or when I happen to re-flash a unit but forgot to delete it in the app first. Then you're stuck if you can't connect to the node and want to re-start/re-use setting up a specific address.

Luckily the new import/export feature now lets me edit my network "offline", so I can manage, altough it's not as smooth a process as it used to be :-)

BR,
Pär-Olof

roshanrajaratnam, 2019-02-06:
@POHakansson I can understand your point and yes the security aspects are quite important here.

However have you looked at our latest sdk 3.1.0 which seems to work much better when it comes this assertion error. I see it very rarely so give it a go and if you see this please open a ticket on Devzone and our tech support team will be happy to  help you.