nordicsemi/Android-nRF-Mesh-Library #147: Network config JSON files must be named same as meshUUID

POHakansson, 2019-01-31:
Describe the bug
When trying to load a JSON network config file, the file name must be identical to the meshUUID inside the file. Very annoying if you're trying to edit the file and happen to call it <something_else>.json :-)

To Reproduce
Steps to reproduce the behavior:
1) Change the name of a valid network config file to a name which is not <meshUUID>.json
2) Try importing it
3) Error message "Invalid file type"

Expected behavior
It should be possible to import a valid config file (also with a human readable name :-)

Platform details:

    Device: Galaxy S7 and Galaxy S8
    OS: Android 8.0.0
    1.1.0 January 16th from Play Store

roshanrajaratnam, 2019-02-06:
Hi @POHakansson, the library does not validate against a file name, It just imports any network information named with any file name. Just make sure that the JSON file is one that is exported by the nRF Mesh App.

roshanrajaratnam, 2019-02-08:
We could have used the network name but the reason for using the UUID as the name is to avoid overwriting files in case someone forgot to change the network name after resetting a network. I can always prompt the user to put in a desired name though.

closed 2019-03-28 by roshanrajaratnam