nordicsemi/Android-nRF-Mesh-Library #153: wrong parsing of 3 byte OpCodes

HaukeRa, 2019-02-03:
---

**Describe the bug**
`MeshParserUtils` line 478:
```
 return ((byte) (accessPayload[0] & 0xFF) | (byte) ((accessPayload[1] << 8) & 0xFF) | (byte) ((accessPayload[2] << 16) & 0xFF));
```
& 0xFF after shifting (and then casting to byte) yields always 0. The calculation above always results in the returned opCode always being equal to the first element of the accessPayload only, both bytes after that are effectively discarded.

**Expected behavior**
A clear and concise description of what you expected to happen.
Applying & 0xFF before shifting results in the desired result as the bytes are converted to an equivalent int without extending the signed bits.
```
 (accessPayload[0]&0xFF ) | (((accessPayload[1]&0xFF) << 8) ) | (((accessPayload[2]&0xFF) << 16));
```

HaukeRa, 2019-02-03:
Also, the order is twisted. The Mesh specification states in table 3.43, that the byte at position 0 is the most significant byte, the current implementation uses it as the least significant byte

roshanrajaratnam, 2019-02-05:
Thanks for pointing this out. I have fixed this internally.

PR #155 (2019-02-12): Improvements/add group controlling #129
* Fixes #129,  #41 
  - Replaces the scanner view with a group view. In order to scan/provision a device, use the add floating action button.
  - Group view adds support for creating/editing a group. When subscribing a model to a group a user may create or select an existing group
  - Model Configuration view now only sends messages to unicast addresses.
* Fixes #128 
  - The app uses the provisioner sequence number as the tid
* Fixes #153
  -  Fixed invalid endianness 
* Fixes #140 
  - Fixed updating a hardcoded key
* Fixes #139 
  - Added missing `not [... cut]
closed 2019-02-05 by roshanrajaratnam