nordicsemi/Android-nRF-Mesh-Library #154: Provisioning with preshared key and select input/output OOB

DomiR, 2019-02-06:
**Is your feature request related to a problem? Please describe.**
We would like to be able to select the oob type and also support preshared keys.

**Describe the solution you'd like**
Implement the api for this. 

I'm starting working on this today and I wanted to get some feedback first. I would like to get access to the `ProvisionCapabilities` on `onProvisioningStateChanged` but I currently only get the raw data. Should we make the parser public or add another callback with the ProvisioningCapabilites and call both callbacks.

I think I should be able to add another `startProvisioningWithPresharedKey(String key)` and `startProvisioningWithInputOOB(Integer random)` and see where it goes. I should be able to test this with our devices. We also have some buttons here, so testing inputOOB should also be possible.

roshanrajaratnam, 2019-02-06:
Hi @DomiR sounds interesting. The `ProvisioningCapabilities` are stored in the `UnprovisionedMeshNode` so you can call `getProvisioningCapabilities()` to find out the capabilities of a node.

It would be nice to implement this part as we haven't really spent time looking in to this as we were focusing on getting the basic mesh implementation out. Please do suggest any improvements wherever you think is appropriate in the provisioning part of of the library as well.

DomiR, 2019-02-06:
For reference: #123

DomiR, 2019-02-15:
Implemented in #157

closed 2019-03-05 by roshanrajaratnam