nordicsemi/Android-nRF-Mesh-Library #165: Can App subscribe to an address like a device?

kcjian, 2019-02-28:
**Is your feature request related to a problem? Please describe.**
When my App accesses the network, I want to receive the release information of the device. I want to ask if the App can subscribe to an address?

**Describe alternatives you've considered**
Implementing this functionality through gateways and servers

roshanrajaratnam, 2019-02-28:
@kcjian Are you saying that you want to get the publication settings that's currently set on a node?

kcjian, 2019-02-28:
@roshanrajaratnam 
Like mutual subscriptions between devices, I want to subscribe to devices with my mobile phone.
For example, there are two sensors A and B. When the data of A changes, they will publish information to address 0xAAAA. When the data of B changes, they will publish information to address 0xBBBB. I want to subscribe to these two addresses (0xAAAA, 0xBBBB) with my mobile phone to receive these information.

roshanrajaratnam, 2019-02-28:
Right, this is something you can do by setting a proxy filter on one of the nodes you are connected. You can turn on the white list filter on one of the nodes and then add the two addresses you wish to receive the data from. Also note that once you disconnect from the node you will lose the proxy filter settings because its per connection and is cleared upon disconnection by the node itself.

kcjian, 2019-02-28:
I see. Thank you very much.

roshanrajaratnam, 2019-02-28:
@kcjian Happy to help! I will close this issue and if have anymore questions about this please reopen again.

closed 2019-02-28 by roshanrajaratnam