nordicsemi/Android-nRF-Mesh-Library #167: BaseMeshMessageHandler has lots of redundant code

HaukeRa, 2019-03-01:
---
BaseMeshMessageHandler has a lot of branches that discriminate based on the type of message sent, but all of the branches basically do the same.

I checked on some of the classes (xxxState) and found no changes in control flow elsewhere based on the actual instance type. Is there a good reason for all the branches? Otherwise I propose to generalize the code here to make it easier to read, less verbose and easier to extend.

The reason why I'm working on this is that I try to pull all the (non provisioning related) message handling out of the provisioning library. While implementing the messages in the provisioner lib makes sense to have a library that can "do something", they are strictly not really related to the provisioiner functionality. I currently implement a more formalized way of using client and server models within the provisioner device (to make it a node within the network finally). Relates to #151 .

HaukeRa, 2019-03-02:
If this is actually the case and there is now conditional execution based on the actual type of the states, then all the subclasses of GenericMessageState are redundant, as they all do exactly the same except of logging a different message during `executeSend()` which definitely is not worth the redundancy

roshanrajaratnam, 2019-03-04:
The intial design of the library did not have its own classes for messages and it was the states that did handle generated the messages itself. However this is something in my backlog to clean up the state classes redundant extensions of the state classes and just leave the base state itself.

HaukeRa, 2019-03-04:
Actually, the changes needed are not large at all. I've done it in my local dev branch.

roshanrajaratnam, 2019-03-04:
Of course not.

PR #181 (2019-03-12): Improvements/remove message states #167
* Fixes #167 
  - Removes the unnecessary states
closed 2019-08-06 by roshanrajaratnam