nordicsemi/Android-nRF-Mesh-Library #168: BouncyCastleProvider is installed in an error-prone way

HaukeRa, 2019-03-02:
---

**Describe the bug**
The installing of the Security provider is done within a static block in SecureUtils.java:
`    static {
        Security.insertProviderAt(new org.spongycastle.jce.provider.BouncyCastleProvider(), 1);
    }`
This is very error prone. There is no guaranty when the static block is executed so it can happen that one uses security functionality before the provider is inserted. For the static block to be executed, a code path that actually uses a SecureUtils.java must be executed.

Currently that happens somehow somewhere, but I refactored a little and ran into the error, that during `ProvisioningPublicKeyState#generateKeyPairs()`, the BouncyCastleProvider was not yet installed.

**Expected behavior**
This is a good example why static is "bad". To make things more clear I would move it into a static method that is called early on (for example in a static block in MeshManagerApi, or even better in the constructor)

roshanrajaratnam, 2019-03-04:
@HaukeRa Thanks for reporting this, I haven't seen this behavior the bouncy castle gets called during the provisioning state or either when sending a generating a mesh message. It does makes sense to not have it set up in a non-static way.

HaukeRa, 2019-03-04:
I haven't seen this behaviour previous to some refactoring, too. It just shows that an "unrelated" change can break something somewhere else. The Security.insertProviderAt method just does nothing if you install the same provider at the same position twice, so it should be safe to invoke it from the constructor of MeshManagerApi

roshanrajaratnam, 2019-03-04:
Okies, let's go with the safer option then!

PR #171 (2019-03-05): Improvements/fix node list ui
* Fixes #169 
  - Added an additional api to override default 5 second attention timer.
`identifyNode(@NonNull final UUID deviceUuid, final String nodeName, final int attentionTimer)`
* Fixes #168 
  - Moved bouncy castle insertion to MeshManagerApi.
* Updated node list UI to match groups list.
* Improvements made to ManageAppKeysActivity. This was listing app keys that were not added although the binding will fail as expected. Now the activity lists only app keys that are added for binding, making it easier for the user.
* Removed unused resources and strings.
PR #172 (2019-03-05): Release v1.2.0
* Fixes #143,  
* Fixes #129, #41 
  - Adds support for controlling groups. Previously this was done in each model configuration view but now model configuration will only allow sending messages to the unicast address itself.
* Fixes #159 
  - Fixes a documentation bug after renaming callbacks.
* Fixes #149 
  - All addresses now use ints instead of byte arrays.
* Fixes #134 
  - Fixes a bug relating to broken parcelable implementations
* Fixes #169 
  - Added an additional api to override default 5 second attention timer.
`identifyNode(@NonNull final UUID deviceUuid, final Stri [... cut]
closed 2019-03-05 by roshanrajaratnam