nordicsemi/Android-nRF-Mesh-Library #169: Expose attentionTimer as duration.

bangonkali, 2019-03-04:
**Is your feature request related to a problem? Please describe.**
In the iOS Version of the library it seems the [`attentionTimer` is exposed as `duration`](https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/e2ce5afbb89b2200c659927d3634dd1c0a0bdba5/nRFMeshProvision/Classes/Provisioner/ProvisioningStates/InviteProvisioningState.swift#L26-L35).

```swift
    required init(withTargetNode aNode: UnprovisionedMeshNodeProtocol) {
        target                              = aNode
        duration                            = 0x05 //Default attention timer is 5 seconds
        super.init()
        target.basePeripheral().delegate    = self
        //If services and characteristics are already discovered, set them now
        let discovery                       = target.discoveredServicesAndCharacteristics()
        provisioningService                 = discovery.provisionService
        dataInCharacteristic                = discovery.dataInCharacteristic
        dataOutCharacteristic               = discovery.dataOutCharacteristic
    }
```

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

Is it possible we also expose the same in Android library? It seems this one is [hard coded](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/a6857e47752cfc667e0601eca1f0bec6b8d97da0/android-nrf-mesh-library/meshprovisioner/src/main/java/no/nordicsemi/android/meshprovisioner/MeshProvisioningHandler.java#L287-L294).

```
    private void sendProvisioningInvite(final UnprovisionedMeshNode unprovisionedMeshNode) {
        isProvisioningPublicKeySent = false;
        isProvisioneePublicKeyReceived = false;
        attentionTimer = 0x05;//0x0A;
        final ProvisioningInviteState invite = new ProvisioningInviteState(unprovisionedMeshNode, attentionTimer, mInternalTransportCallbacks, mStatusCallbacks);
        provisioningState = invite;
        invite.executeSend();
```

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

It would be interesting to make this user configurabl on the client side.

**Additional context**
Add any other context or screenshots about the feature request here.

`n/a`

roshanrajaratnam, 2019-03-04:
@bangonkali this is correct and the attention timer will be exposed.

roshanrajaratnam, 2019-03-04:
@bangonkali Fixed internaly

PR #171 (2019-03-05): Improvements/fix node list ui
* Fixes #169 
  - Added an additional api to override default 5 second attention timer.
`identifyNode(@NonNull final UUID deviceUuid, final String nodeName, final int attentionTimer)`
* Fixes #168 
  - Moved bouncy castle insertion to MeshManagerApi.
* Updated node list UI to match groups list.
* Improvements made to ManageAppKeysActivity. This was listing app keys that were not added although the binding will fail as expected. Now the activity lists only app keys that are added for binding, making it easier for the user.
* Removed unused resources and strings.
PR #172 (2019-03-05): Release v1.2.0
* Fixes #143,  
* Fixes #129, #41 
  - Adds support for controlling groups. Previously this was done in each model configuration view but now model configuration will only allow sending messages to the unicast address itself.
* Fixes #159 
  - Fixes a documentation bug after renaming callbacks.
* Fixes #149 
  - All addresses now use ints instead of byte arrays.
* Fixes #134 
  - Fixes a bug relating to broken parcelable implementations
* Fixes #169 
  - Added an additional api to override default 5 second attention timer.
`identifyNode(@NonNull final UUID deviceUuid, final Stri [... cut]
closed 2019-03-05 by roshanrajaratnam