nordicsemi/Android-nRF-Mesh-Library #170: Individual controls controlling multiple elements

SamS3lly, 2019-03-04:
**Describe the bug**
Not sure if this is a bug or it should work like this, summary: having 2 or more nodes, if we configure both/all elements and models to the same group address and then control one of the elements with the on/off button or the generic level slider, all nodes subscribed to the same group with change their state as well, is this suppose to happen?

**To Reproduce**
Steps to reproduce the behavior:
provision 2 or more devices
configure the same group address on the same models on all the elements (ie on/off model or generic level model)
turn on / off one of the elements or change the slider
all the elements configured to subscribe to the same group address will turn on / off

**Expected behavior**
shouldn't only the individual element be affected with the on / off? the only time that all the elements should respond to the on / off command is when sending that command to the group address?

**Platform details:**
 - Device: Honor 6x
 - OS: Android 7
 - Library Version (or Commit) 1.1.0

roshanrajaratnam, 2019-03-04:
This is something to do with how the app is implemented because there is no other view to control groups. The problem being the app sends the message to group addresses if that model has susbcribed to any.

However this behaviour is  different on the latest thetest dev branch. The model configuration view now only sends to the unicast address and a separate group view has been implemented to control groups.

closed 2019-03-04 by roshanrajaratnam

SamS3lly, 2019-03-04:
@roshanrajaratnam 
Thank you for the quick reply! Another question we're wondering is:

We are implementing the groups into the UI already and we have an open question, when adding a new item to the group, should this "item" be a whole element with all its models? or should every model be added separately? let's say that we have a server with an on/off model and generic level model, should both models be added to the group automatically or should every model be added separately?

URL: https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/issues/117

roshanrajaratnam, 2019-03-05:
@SamS3lly please do not open/create issues relating to other repos, it will only cause confusion to other users.

zanibonilighting, 2019-03-05:
Hi! I would like to help my colleague a bit here :)
You are right, we shouldn't mix up repos questions, but the functionality relates to both, Android and iOS, but still is not good to mix different questions even if they are related, so I will post another separate question. Thanks!

roshanrajaratnam, 2019-03-06:
@zanibonilighting Thanks for understanding and we are happy to help and its easier to keep track of :)