nordicsemi/Android-nRF-Mesh-Library #174: when adding items to groups, should be whole elements with all its models or individual models?

zanibonilighting, 2019-03-05:
**Is your feature request related to a problem? Please describe.**
More than a feature request is more a question regarding the specification / general implementation.
A quick summary, we ran the current Android groups implementation with few test users (friendly clients) and they found it a bit confusing and nonintuitive.

**Describe the solution you'd like**
They asked if the groups configuration could be simplified as:

* Create a group, ask for name (suggest generic name for fast deployments, ie Group 1)
* Set Group address automatically to new group
* Show "add new item" (nodes? items? elements? models?)
* Display list of nodes on the network (Nodes as the first element of the node which is the first addressable item)
* When click an node, display the list of elements

Here is the question, should all the models inside an element be added to the group? or there should be another step as:

* Click on the element, display the list of models
* Select the models to be added to the group
* Configure group address and apply publishing (Clients) and subscribing (Servers) models

**Describe alternatives you've considered**
Should all the models on the element be added to the group automatically or each model inside an element should have its own configuration?

roshanrajaratnam, 2019-03-06:
@zanibonilighting Grouping can be a bit confusing. One of the things we thought about was going through all the elements and selecting the models can be a bit confusing for a user as well because there could be a lot of elements and models within a node. In addition automating this process of subscribing the models to a group can be a bit cumbersome because messages are not guaranteed to be delivered and can get lost or dropped. This is one of the reasons we just allowed uses to subscribe to existing groups from the model views.

To answer your question no only the specified models should be added to the group otherwise you will be congesting your network as well. So IMHO the user has to pick the specific model that should subscribe to a group. I would actually do this in the background while the user pick which model is to be subscribed to a group.

closed 2019-03-19 by zanibonilighting

roshanrajaratnam, 2019-03-20:
@zanibonilighting sorry I missed out on this issue. it's nice that its automated and I like this approach that you get to select the keys and then the models and they will be bound automatically!