nordicsemi/Android-nRF-Mesh-Library #178: Publish Period settings do not work

kcjian, 2019-03-06:
[image]

 hi，according to the Mesh protocol I think
 **int period=(resolution << 6) | steps ;**
but in the [ConfigModelPublicationSet.java]( https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/master/android-nrf-mesh-library/meshprovisioner/src/main/java/no/nordicsemi/android/meshprovisioner/transport/ConfigModelPublicationSet.java),Line 185
**paramsBuffer.put((byte) ((publicationSteps << 6) | publicationResolution));**
I hope you can have a look.

**Platform details:**
 - Library Version 1.2.1

roshanrajaratnam, 2019-03-11:
@kcjian Thanks for reporting this. It is broken!

kcjian, 2019-03-11:
I found this problem in many places, such as SceneRecall,GenericOnOffSet, looking forward to your updates.

roshanrajaratnam, 2019-03-11:
I have merged this to dev branch now and it should work for you!

kcjian, 2019-03-11:
@roshanrajaratnam Thank you ~

PR #180 (2019-03-11): Bugfix/publish period #178
* Fixes #178
closed 2019-03-26 by roshanrajaratnam