nordicsemi/Android-nRF-Mesh-Library #196: Responding from vendor model longer than 8 bytes

ximplarandy, 2019-05-08:
---

**Describe the bug**
I have a vendor model, and I use the mesh app to send data to the server element.  With a status reply of fewer or equal to 8 bytes (when the message is not segmented).  However, longer messages which are sent in segments, would output an error message "Segment received is from a different src than the one we are processing, let's drop it" would appear, and no message is decoded.  

The part of code in NetworkLayer.java in question is between line 498  and 501:
`             if (src != mMeshNode.getUnicastAddress()) {
                    Log.v(TAG, "Segment received is from a different src than the one we are processing, let's drop it");
                    return null;
                }
`
After removing, the status reply is properly received.  Is this condition needed?  And what is the purpose?

Debugging shows src is D3 and mMeshNode.getUnicastAddress() is D6 (same hardware, different elements)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Node -> One of the vendor server element'
2. Send data with 0x112233445566778899 (longer than 8 bytes)
3. See error "Segment received is from a different src than the one we are processing, let's drop it"

roshanrajaratnam, 2019-05-08:
@ximplarandy Could you provide your specific vendor model example I can test against?

ximplarandy, 2019-05-09:
@roshanrajaratnam , I need to get approval for sending the files.  However, the issue is not really about any specific model, but rather the few lines that I pointed out.  I would guess if I am reading from a model at the top, the two unicast addresses would be the same, and hence no problem.   I am wondering what the purpose of the statements.

roshanrajaratnam, 2019-05-09:
Well the current dev branch should have ideally had removed this lines because now the library can process messages from different addresses simultaneously. Therefore these lines can be removed.

ximplarandy, 2019-05-09:
thanks! Removed and everything works.

closed 2019-05-09 by ximplarandy