nordicsemi/Android-nRF-Mesh-Library #198: Question

SachienRaj, 2019-05-10:
In the MeshProvisionerActivity, i want to invoke the "mViewModel.getMeshManagerApi().startProvisioningWithStaticOOB(node);" method when i start the activity. Somehow I modified the code to invoke the method, but a ClassCastException is thrown (java.lang.ClassCastException: no.nordicsemi.android.meshprovisioner.provisionerstates.ProvisioningPublicKeyState cannot be cast to no.nordicsemi.android.meshprovisioner.provisionerstates.ProvisioningCapabilitiesState). Is it possible to invoke the "mViewModel.getMeshManagerApi().startProvisioningWithStaticOOB(node);" method on start of the activity?

roshanrajaratnam, 2019-05-13:
Hi @SachienRaj I suppose you are trying to provision a device? If that's the case you have to call identify node first and then call start provision after.

closed 2019-05-23 by roshanrajaratnam

EnMMSegur, 2019-10-28:
I have the same problem even though I already identified the node first

EnMMSegur, 2019-10-28:
> I have the same problem even though I already identified the node first

To add, my ProvisioningCharacteristic remain null no matter what i do;