nordicsemi/Android-nRF-Mesh-Library #203: 'sendGenericOnOffGet' doesn't call 'updateMeshMessage' when invoked from onCreate

MeenalBahua, 2019-05-29:
---

When i tried to invoke sendGenericOnOffGet from onCreate method in GenericOnOffServerActivity. it never invokes updateMeshMessage. If we invoke sendGenericOnOffGet through a click event it works fine. During debug or if i add some delay before invoking the the get call, it works fine. What could be the possible reason for this behaviour?

roshanrajaratnam, 2019-05-29:
@MeenalBahua may be you send this before the observer is set up the observer?

MeenalBahua, 2019-05-29:
No, Observer is called in the base class (BaseModelConfigurationActivity) itself.

roshanrajaratnam, 2019-05-29:
Yes that's right, However have you checked if you get a reply from the device at all?

MeenalBahua, 2019-05-29:
Yes, We do receive reply from the device, and i traced its path till 'parseNotifications' in MeshManagerApi. It also But after this it looses its reference. 'parseMeshMsgNotifications' from BaseMeshMessageHandler never gets called. The 'unsegmentedPdu' is of the type 'PDU_TYPE_NETWORK', so it enters the correct switch case as well.

MeenalBahua, 2019-05-29:
Resolved it. There was issue due to multiple observers for same SingleLiveData object.

roshanrajaratnam, 2019-05-29:
Great, was this happenning in the original app or did it happen after your changes?

MeenalBahua, 2019-05-30:
Yes. To reproduce it, call the get method from onCreate in any of the model activities. The observer callback goes to previous activity (nodeconfiguration) instead of the model activity(like GenericOnOffActivity). I changed mMeshMessageLiveData in NrfMeshRepository from SingleLiveEvent to MutableLiveData.

roshanrajaratnam, 2019-05-30:
Right I see, but making it mutable live data will notify all observers including the nodeConfiguration and also will update every time you rotate the screen. May be you can remove the observer onStop() and set it again in onStart()

MeenalBahua, 2019-05-30:
Ok, OnStop gets called a little after next activity's OnCreate. OnPause was a better place to remove observer.

closed 2019-05-30 by MeenalBahua