nordicsemi/Android-nRF-Mesh-Library #204: Scanning Provisioned  Node  Went Wrong

WangManAndroid, 2019-05-29:
2019-05-29 16:20:27.086 25987-25987/no.nordicsemi.android.nrfmeshprovisioner W/System.err: java.lang.NullPointerException: Attempt to get length of null array
2019-05-29 16:20:27.087 25987-25987/no.nordicsemi.android.nrfmeshprovisioner W/System.err:     at org.spongycastle.crypto.params.KeyParameter.<init>(KeyParameter.java:13)
2019-05-29 16:20:27.088 25987-25987/no.nordicsemi.android.nrfmeshprovisioner W/System.err:     at no.nordicsemi.android.meshprovisioner.utils.SecureUtils.encryptWithAES(SecureUtils.java:388)
2019-05-29 16:20:27.088 25987-25987/no.nordicsemi.android.nrfmeshprovisioner W/System.err:     at no.nordicsemi.android.meshprovisioner.utils.SecureUtils.calculateHash(SecureUtils.java:378)
2019-05-29 16:20:27.089 25987-25987/no.nordicsemi.android.nrfmeshprovisioner W/System.err:     at no.nordicsemi.android.meshprovisioner.MeshManagerApi.nodeIdentityMatches(MeshManagerApi.java:702)
2019-05-29 16:20:27.089 25987-25987/no.nordicsemi.android.nrfmeshprovisioner W/System.err:     at no.nordicsemi.android.nrfmeshprovisioner.viewmodels.ScannerRepository.checkIfNodeIdentityMatches(ScannerRepository.java:344)
2019-05-29 16:20:27.089 25987-25987/no.nordicsemi.android.nrfmeshprovisioner W/System.err:     at no.nordicsemi.android.nrfmeshprovisioner.viewmodels.ScannerRepository.access$600(ScannerRepository.java:55)
2019-05-29 16:20:27.089 25987-25987/no.nordicsemi.android.nrfmeshprovisioner W/System.err:     at no.nordicsemi.android.nrfmeshprovisioner.viewmodels.ScannerRepository$1.onScanResult(ScannerRepository.java:91)
2019-05-29 16:20:27.089 25987-25987/no.nordicsemi.android.nrfmeshprovisioner W/System.err:     at no.nordicsemi.android.support.v18.scanner.BluetoothLeScannerCompat$ScanCallbackWrapper$2.run(BluetoothLeScannerCompat.java:320)
2019-05-29 16:20:27.090 25987-25987/no.nordicsemi.android.nrfmeshprovisioner W/System.err:     at android.os.Handler.handleCallback(Handler.java:873)
2019-05-29 16:20:27.090 25987-25987/no.nordicsemi.android.nrfmeshprovisioner W/System.err:     at android.os.Handler.dispatchMessage(Handler.java:99)
2019-05-29 16:20:27.090 25987-25987/no.nordicsemi.android.nrfmeshprovisioner W/System.err:     at android.os.Looper.loop(Looper.java:207)
2019-05-29 16:20:27.090 25987-25987/no.nordicsemi.android.nrfmeshprovisioner W/System.err:     at android.app.ActivityThread.main(ActivityThread.java:6867)
2019-05-29 16:20:27.090 25987-25987/no.nordicsemi.android.nrfmeshprovisioner W/System.err:     at java.lang.reflect.Method.invoke(Native Method)
2019-05-29 16:20:27.090 25987-25987/no.nordicsemi.android.nrfmeshprovisioner W/System.err:     at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:547)
2019-05-29 16:20:27.090 25987-25987/no.nordicsemi.android.nrfmeshprovisioner W/System.err:     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:876)

`
 @Override
    public boolean nodeIdentityMatches(@NonNull final ProvisionedMeshNode meshNode, @NonNull final byte[] serviceData) {
        final byte[] advertisedHash = getAdvertisedHash(serviceData);
        //If there is no advertised hash return false as this is used to match against the generated hash
        if (advertisedHash == null) {
            return false;
        }

        //If there is no advertised random return false as this is used to generate the hash to match against the advertised
        final byte[] random = getAdvertisedRandom(serviceData);
        if (random == null) {
            return false;
        }

        //if generated hash is null return false
        final byte[] generatedHash = SecureUtils.calculateHash(meshNode.getIdentityKey(), random, AddressUtils.getUnicastAddressBytes(meshNode.getUnicastAddress()));

        return Arrays.equals(advertisedHash, generatedHash);
    }
 `
 reason :meshNode.getIdentityKey()  is null

WangManAndroid, 2019-05-29:
when we relaunch  app  i found  the  filed  identityKey   is null

roshanrajaratnam, 2019-05-29:
Thanks for reporting this, I'll take a look and get back to you on this.

roshanrajaratnam, 2019-07-10:
Hi @WangManAndroid sorry for the delay, yes this is certainly a bug in the library. Currently the identity key is not stored as a property of the mesh node. Howeer it should be part of the network key because its derived using a network key. await a fix!

PR #212 (2019-07-15): Improvements/update settings ui
* Multiple Provisioenr Management
  - Add,Edit and Remove Provisioners
  - Allocate Unicast/Group/Scene ranges
* Advanced Key Management
  - Add/Edit/Remove Network/Application keys
  - Bind Application keys to a network key
* Creating groups with virtual addresses
* Sending messages to virtual addresses
* Improvements to Publication Settings
* Fixed incorrect endianness of addresses fixed in ProxyConfigRemoveAddressFromFilter
* Rename elements
* Improvements to Network Import/Export
* UI Improvements
  - Users are now allowed to view all configurations without  [... cut]
closed 2019-08-06 by roshanrajaratnam