nordicsemi/Android-nRF-Mesh-Library #207: Light CTL Temperature Get not working on Android

MeenalBahua, 2019-06-24:
Light CTL Temperature Get message does not return relevant Status message. 
 I implemented the Light CTL Temperature Get & status messages inline with the other Models already implemented. But when i invoke send message, it never invokes the response status message.

While debugging, I have observed that when the response message is received, it comes till "private AccessMessage parseAccessMessage" method in "NerworkLayer.java" and displays a message in logs "Segment received is from a different src than the one we are processing, let's drop it" multiple times, but doesn't go ahead from here. Am i missing something in my "Get message"? I have checked that the relevant Opcodes are being applied.

Let me know in case you need further details.

roshanrajaratnam, 2019-06-24:
Hi @MeenalBahua Did you add the status message opcode [here](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/master/android-nrf-mesh-library/meshprovisioner/src/main/java/no/nordicsemi/android/meshprovisioner/transport/DefaultNoOperationMessageState.java#L66)? Could you also log the source addresses and see if you are receiving any messages from a different node at the same time? Current master branch may fail to process messages in case a message from another node was received at the same time. However this is fixed in the dev branch.

MeenalBahua, 2019-06-24:
Yes, the Opcode is added in the respective file. Also, I have only one node i am using for development. And the message received is from the same node. Though in othet case of other all othet Models/opcodes, "if (isSegmentedMessage(decryptedNetworkPayload[2]))" returns false, but in case of CTL Temperature, it returns True.

roshanrajaratnam, 2019-06-24:
It depends on the device you are communicating with because some devices may send it as a segmented message for extra security and also if the message is too long to be sent in one message. Could you pull the changes from the latest current dev branch to see if it fixes your issue?

MeenalBahua, 2019-06-24:
Let me check.

MeenalBahua, 2019-06-24:
Here's whats happening:

In the current setup, only CTL Temperature Status message is segmented. When we receive a segmented message, It checks if the src address of received message is same as the nodes unicast address. Unicast address for a node in android app is referenced by address of 1st Element in that node(Say 0001). Now when we click on CTL Temrature server, it is 2nd Element in the node(with Unicast address 0002). So 'if (src != mMeshNode.getUnicastAddress())' returns 'true' in 'NetworkLayer.java', it drops the packet altogether.

roshanrajaratnam, 2019-06-25:
@MeenalBahua Right I see, actually this line is not needed anymore since I have implemented the concurrent processing of messages and this means each address will have its own mesh layers. Thanks for debugging this for me. if you want o contribute towards the repo you can also send us a pull request.

WangManAndroid, 2019-06-28:
`  public static boolean isSrcInNode(ProvisionedMeshNode node,int src){
      List<Element>elements=new ArrayList<>(node.getElements().values()) ;
      if (elements.isEmpty())return true;
      boolean dstInNode=false;
      for (int i = 0; i <elements.size() ; i++) {
          String src1=MeshAddress.formatAddress(elements.get(i).getElementAddress(),false);
          String src2=MeshAddress.formatAddress(src, false);
          if (src1.equals(src2)){
              dstInNode=true;
          }
      }
        return dstInNode;
  }
}`

roshanrajaratnam, 2019-06-28:
@MeenalBahua did you have any luck with this?

MeenalBahua, 2019-06-28:
Yes. In the current src code I added the following check for elements, which seems to working fine as of now:

`//Check if the received segmented message is from the same src as the previous segment
                if (src != mMeshNode.getUnicastAddress()) {
                    ///Code added to check unicast addr for all elements for a node : Meenal
                    Map<Integer, Element> tempElements = mMeshNode.getElements();
                    boolean matchAddr = false;
                    for(Map.Entry<Integer, Element> entry : tempElements.entrySet()) {
                        Element value = entry.getValue();
                        if(src == value.getElementAddress()) {
                            matchAddr = true;
                            break;
                        }
                    }
                    if(!matchAddr) {
                        Log.v(TAG, "Segment received is from a different src than the one we are processing, let's drop it");
                        return null;
                    }
                }`

roshanrajaratnam, 2019-07-01:
@MeenalBahua Ok great sounds great. In my opinion, if you're using the current dev version of the library, you can remove that check completely.

MeenalBahua, 2019-07-01:
Currently i am using the master version for the lib and its already into UAT. Though at some point i will have to integrate with updated dev version. By when do you plan to merge the current dev into master?

roshanrajaratnam, 2019-07-01:
I am nearing a release, so probably either later today or tomorrow.

MeenalBahua, 2019-07-01:
Ok. I guess i can plan something later next week to merge the new master into my current app.

MeenalBahua, 2019-08-01:
I was wondering if & when you will be merging the current dev branch changes in the master branch? Or if the dev branch good enough to use for long term development?

roshanrajaratnam, 2019-08-01:
@MeenalBahua sorry it took a little longer than expected. However we are quite close to a release like I said last time ;)

In my opinion dev is good enough for long term development.

PR #212 (2019-07-15): Improvements/update settings ui
* Multiple Provisioenr Management
  - Add,Edit and Remove Provisioners
  - Allocate Unicast/Group/Scene ranges
* Advanced Key Management
  - Add/Edit/Remove Network/Application keys
  - Bind Application keys to a network key
* Creating groups with virtual addresses
* Sending messages to virtual addresses
* Improvements to Publication Settings
* Fixed incorrect endianness of addresses fixed in ProxyConfigRemoveAddressFromFilter
* Rename elements
* Improvements to Network Import/Export
* UI Improvements
  - Users are now allowed to view all configurations without  [... cut]
closed 2019-08-06 by roshanrajaratnam