nordicsemi/Android-nRF-Mesh-Library #210: Disabling Relay

BubblyNetDev, 2019-06-27:
---

**Describe the bug**
When turning off the enable relay toggle the node does not change.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Configuration server
2. Click on set relay button
3. Switch toggle off
4. See error

**Expected behavior**
When the relay is off the node is no longer a relay.

**Platform details:**
 - Device: Galaxy Tab S5E
 - OS: Android 9.0
 - Library Version (dev branch)

**Logs / Screenshots**

[image]

PM-BubblyNet, 2019-06-27:
hi!

We found a toggle on the configuration server model that seems to be intended to enable/disable the relay functionality, we tested but its not really doing nothing on the node I believe?

Then we found https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.meshsdk.v3.1.0%2Fmd_MIGRATION_GUIDE.html on the Migration from v2.0.1 to v2.2.0 it says: Note that the transport-layer options are still only present on the old nrf_mesh_opt API. They will be migrated in the next release. 

Does this means that 3.1.0 has this transport-layer options already?

The link to the question on the DevZone / SDK is: https://devzone.nordicsemi.com/f/nordic-q-a/48343/relay-nodes-on-mesh-sdk

Thanks!

roshanrajaratnam, 2019-06-27:
@BubblyNetDev Please don't create duplicate questions both on Devzone and Github this will only cause confusion for all users. However our tech support will get back to you on this.

closed 2019-06-27 by roshanrajaratnam

PM-BubblyNet, 2019-06-27:
Confusion? why? this github is for the Android provisioning library and the question on DevZone is about the SDK, they are not duplicated questions but 2 separate questions related to each other, and when this happens its good practice and very helpful to link the questions.
For example, when I asked the question on the DevZone couple weeks ago it wasn't pointed out that the Android Library already had a toggle that seems intended for this purpose...?
The questions remains, is there a way to enable / disable the relay functionality of the node on the Android Library?