nordicsemi/Android-nRF-Mesh-Library #214: Connection failure after provisioning

dominicbar, 2019-07-18:
2019-07-18 20:26:08.093 23990-23990/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: registerApp()
2019-07-18 20:26:08.094 23990-23990/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: registerApp() - UUID=c5d4d7d1-72c1-433c-b4f6-dcf38145ad0c
2019-07-18 20:26:08.098 23990-24018/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: onClientRegistered() - status=0 clientIf=7
2019-07-18 20:26:08.636 23990-24018/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=7 device=F9:F6:35:68:6F:E6
2019-07-18 20:26:08.636 23990-24018/no.nordicsemi.android.nrfmeshprovisioner V/NrfMeshRepository: Disconnected
2019-07-18 20:26:08.657 23990-24018/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: close()
2019-07-18 20:26:08.657 23990-24018/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: unregisterApp() - mClientIf=7
2019-07-18 20:26:08.659 23990-24018/no.nordicsemi.android.nrfmeshprovisioner E/NrfMeshRepository: Error: Error on connection state change Error Code: 133 Device: F9:F6:35:68:6F:E6

One issue is we are not finding provisioned node,(sorry i cant provide u logs)
Second isssue is provision node found and connection fails(logs are for second issue)

roshanrajaratnam, 2019-07-18:
Hi @dominicbar, I see the error code appears to be 133. This usually happens on Android due to a connection timeout that can happen and ble resources not being released. I have added some delays before starting a connection to avoid this as much as possible, but this does not mean that it can never happen. Have you made any changes?

dev-sareno, 2019-07-29:
I've encountered the same issue before but so far it was lessen after upgrading the no.nordicsemi.android:ble from 1.2.0 to 2.1.1

dominicbar, 2019-07-29:
where can i get the 2.1.1?
on playstore it is 1.2.1

roshanrajaratnam, 2019-07-29:
@dev-sareno Right, I'll look in to migrating to BLE library 2.1.1. @dominicbar dev-sareno is referring to the ble library used in the app

roshanrajaratnam, 2019-07-29:
@dominicbar @dev-sareno which mesh sdk are you using?

dominicbar, 2019-07-29:
We are using 1.2.1 SDK

roshanrajaratnam, 2019-07-29:
@dominicbar I am refering to the Mesh SDK on the fw.

dev-sareno, 2019-07-29:
> @dominicbar @dev-sareno which mesh sdk are you using?

me too v1.2.1

dominicbar, 2019-07-29:
@roshanrajaratnam 1.2.1 mesh sdk

dominicbar, 2019-07-29:
One thing we have noticed is that, it works fine for most of the redmi phones, 10 out of 10 times,
but for other phones it works fine 6/10 times

roshanrajaratnam, 2019-07-29:
There is no 1.2.1 version for nRF5 SDK for mesh, what you are referring to is the Android Library for Mesh. It has to be one of the versions listed [here](https://www.nordicsemi.com/Software-and-Tools/Software/nRF5-SDK-for-Mesh/Download#infotabs).

@dominicbar yeah like I said this is mostly an Android issue...What are the other phones you are referring to here? What make and model are they?

dominicbar, 2019-07-29:
@roshanrajaratnam  we are using 
v3.0.0 nRF5 SDK for Mesh Software Development Kit

other phones are 
asus zenphone max pro m1, m2, 
lenovo A700-a,
moto G5,
Samsung J7

xiaofengming, 2019-07-30:
What is your solution?

zhangping001, 2019-09-09:
error 133问题有解决方案吗？悲剧，一直连接不上！

WuLFGit, 2019-09-10:
Hi everyone! i don't know if this may be useful, but i noticed that this error seems to happen only after i already provisioned the first device. From testing, i see that closing and re-opening my app let me provision another node, but then i get error code 133 again for further nodes. Only if i re-start my app over and over i can provision multiple nodes without getting the error. Could it be that after the first provisioning something needs to be refreshed behind the scenes?

Thank you!

WangManAndroid, 2019-09-16:
@xiaofengming   你可以等待入网完毕后，延迟2s然后再连接，不必再次扫描，前提是保存刚才入网的设备BluetoothDevice，这样极大加快后续操作流程

roshanrajaratnam, 2019-10-07:
Closing this issue. Please feel free to re-open if the need arises.

closed 2019-10-07 by roshanrajaratnam