nordicsemi/Android-nRF-Mesh-Library #217: error 133

xiaofengming, 2019-07-29:
133 error when identifyNode and startProvisioning ?

dev-sareno, 2019-07-30:
Similar to [this one](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/issues/214#issue-469830873).

xiaofengming, 2019-07-30:
What is your solution?

dev-sareno, 2019-07-30:
I upgraded the BLE library from 1.2.0 to 2.1.1 as i mentioned [here](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/issues/214#issuecomment-515857878). It doesn't totally fixed the solution but however it lessen the occurrence of 133 error.

xiaofengming, 2019-07-31:
Execute this getNrfMeshRepository().getBleMeshManager().refreshDeviceCache();  before connecting

but it reduced the occurrence of 133 errors.

roshanrajaratnam, 2019-07-31:
@xiaofengming This is already done during every disconnect. Also @dev-sareno the reason for BLE library version 2.1.1 lessens this issue is that it has a retry mechanism implemented.

xiaofengming, 2019-07-31:
Version 2.1.1 is not compatible with your library

roshanrajaratnam, 2019-07-31:
@xiaofengming yes it's in the backlog to be added

zhangping001, 2019-09-09:
mHandler.postDelayed(new Runnable() {
            @Override
            public void run() {
                ConnectRequest connectRequest = mBleMeshManager.connect(bluetoothDevice);
               // mBleMeshManager.internalConnect(bluetoothDevice,connectRequest);//private method !!!!!
            }
        },1000);

in lib 2.1.1,Internal private method, how to connect BLE?

dev-sareno, 2019-09-29:
> ```
>     mHandler.postDelayed(new Runnable() {
>         @Override
>         public void run() {
>             ConnectRequest connectRequest = mBleMeshManager.connect(bluetoothDevice);
>            // [...]

You have to call `connectRequest.enqueue()` for async or  `connectRequest.await()` for sync call.

closed 2019-10-11 by xiaofengming